AlexandreRouma/SDRPlusPlus #399: clang-cl error

gvanem, 2021-10-01:
When using clang-cl to build (instead of MSVC's `cl`), I get this error:
```
sddc_source/src/libsddc/Core/arch/win32/FX3handler.cpp(352,10): error: cannot initialize return object of type 'bool'
      with an rvalue of type 'nullptr_t'
                return nullptr;
                       ^~~~~~~
```

Mixing `nullptr` with `false` !? I fixed it by:
```diff
--- a/sddc_source/src/libsddc/Core/arch/win32/FX3handler.cpp 2021-09-30 11:29:25
+++ b/sddc_source/src/libsddc/Core/arch/win32/FX3handler.cpp 2021-10-02 00:39:50
@@ -349,7 +349,7 @@

        if (!readContext)
        {
-               return nullptr;
+               return false;
        }

```

AlexandreRouma, 2021-10-01:
The SDDC source isn't meant to be used, it's not working. Check [this section of the readme](https://github.com/AlexandreRouma/SDRPlusPlus#sources) for which modules can be built. Modules marked as unfinished should never be built, those in beta can be built but will likely contain bugs and/or miss features.

By the way, the windows version should be built with MSVC and not clang as this will create linking issues with modules distributed as binaries. (Also I don't test for it)

gvanem, 2021-10-02:
> The SDDC source isn't meant to be used, it's not working.

That's okay. Closing.

closed 2021-10-02 by gvanem

gvanem, 2021-10-02:
> By the way, the windows version should be built with MSVC and not clang

Probably if one uses `lld-link.exe` with clang-cl. Using `link.exe` works fine AFAICS for both. 
Besides using clang-cl is handy to detect warnings (possible errors?) like these:
```
core/src/utils/networking.cpp(7,17): warning: 'extern' variable has an initializer [-Wextern-initializer]
    extern bool winsock_init = false;
                ^
```

I'm amazed it's legal in C++.

AlexandreRouma, 2021-10-02:
> I'm amazed it's legal in C++.

I'll check again but I remember checking if it worked and it did.
Also neither GCC or MSVC give a warning about it