AlexandreRouma/SDRPlusPlus #403: Feature Request: Frequency Bookmark Names By Signal Strength Meter

BT-Justice, 2021-10-02:
Thank you very much for SDR++.  This program allowed me to switch to Linux full-time.

I'd like to request a feature.  Instead of bookmark names being displayed on the signal window, could they instead be displayed to the left of the signal strength meter?  If you bookmark all of the USA Marine frequencies, it gets quite crowded when they are displayed on the signal window.

[image]

AlexandreRouma, 2021-10-02:
Hello,

I'm not quite sure what you mean by "to the left of the signal strength meter", could you show an example?
The position of the bookmarks correlate with their frequencies so I'm not quite sure what moving them up there would change.
Maybe adding an option to disable the lines and just keep the text would help? You can also already hide the bookmarks from the signal window by selecting "none" as the "bookmark display mode" parameter

corvus-ch, 2021-10-02:
I have experienced the same issue as @BT-Justice. For me, the issue are not the lines but the bookmark text shown at the top. If bookmarks are close together, their label start to overlap.

I do not see where else to put them but what about displaying the label only when the mouse pinter is over the bookmark line?

BT-Justice, 2021-10-02:
I guess I didn't explain it well.  Instead of the yellow lines and yellow tags for bookmarks in the signal window, have the label shown to the left of the signal strength meter.  Like this:

[image]

BT-Justice, 2021-10-02:
Or maybe in the waterfall like this:

[image]

AlexandreRouma, 2021-10-02:
> I do not see where else to put them but what about displaying the label only when the mouse pointer is over the bookmark line?

Might work but a line is very hard to hover precisely and it's not very intuitive so I'll have to think about it

> I guess I didn't explain it well. Instead of the yellow lines and yellow tags for bookmarks in the signal window, have the label shown to the left of the signal strength meter. Like this:

The issue is this negates the point of having the bookmarks in the first place. From what I understand this would just show the bookmark you're currently tuned to, but since you already had to select it in the list you already know what bookmark you're tuned to. Might still not understand it correctly tho so feel free to correct me on this

BT-Justice, 2021-10-02:
I guess a good comparison would be what SDRuno does when it scans.  It goes through bookmarks which are labeled.  When it stops on an active frequency, the SDRuno Scanner window shows the name of the bookmark.  I guess I am looking at this as if SDR++ already had a scanner.  I know if it does get a scan feature, it is going to look odd if the user has the bookmarks displayed as they are presently.

Or maybe I am saying lose the current bookmark display altogether and have it to where if you are tuned to a frequency that happens to be bookmarked, whatever the user called the bookmark is shown in a area as I pictured in above posts.  I forgot to turn the yellow bookmarking feature off when I did the pictures so maybe that is the confusion.

[image]

shuyuan-liu, 2021-10-02:
A compromise option could be to stagger bookmarks when they are close together, as in gqrx. This gives a bit more room but is likely to clutter the spectrum display when there are too many bookmarks.

[image]

BT-Justice, 2021-10-02:
@shuyuan-liu That is an option, but I think a window showing the name of the bookmark would be preferred if possible.  At least, that is what I am hoping for :)

BT-Justice, 2021-10-02:
Let me try to clarify this some more.

So SDR++ would have a black window at the top of the waterfall:

[image]

And if it was tuned to a frequency that was bookmarked, it would display the name of the bookmark centered in that window.  Like this:

[image]

shuyuan-liu, 2021-10-02:
The staggered approach from gqrx is mainly to make adjacent bookmarks readable and easier to click on, solving @corvus-ch's problem. I'd suggest we open a separate issue for that feature. I think a large text showing the currently tuned bookmark name can co-exist with that. Since the current bookmarks are already togglable in the freq manager module, we don't have to remove them.

I'm not too sure about the black window though, because when it is empty I think the spectrum and waterfall would look too far apart. It could disappear when it's empty but then we would need the waterfall to be able to shift up and down - I don't know how difficult that would be to implement.

BT-Justice, 2021-10-02:
I like the idea of the window above to waterfall, but this would work too between the frequency and the signal strength meter:

[image]

darauble, 2022-06-23:
I have copied Frequency Manager and created my own plugin for exactly this issue:

https://github.com/darauble/bookmark_manager

[image]

closed 2022-09-30 by AlexandreRouma