AlexandreRouma/SDRPlusPlus #404: rtl_tcp samples once then crashes during re-tune

ljkenny, 2021-10-02:
SDR++ appears to connect okay, however the frequency analyser is noisy and still.  The waterfall is mostly red lines.

When I attempt to tune to a different frequently SDR++ crashes without an error message.

SDR++ doesn't seem to take any flags, so I'm not sure how to put it into debug mode for more info.

Installed on Debian Linux via the *.deb from the releases page:

```
$ dpkg -l | grep sdrpp
ii  sdrpp               0.2.5-282           all          Bloat-free SDR receiver software
```

AlexandreRouma, 2021-10-02:
Hard to say what's going on with that information.
What exact OS are you running (aka what version of debian or derivative)? Where is the RTL-TCP server? WAN or LAN? What sample rate is it running at? etc...
Start SDR++ in a terminal and it'll print out the logs, send full logs from start up to crash.
Also send a screenshot of the full UI with the source menu in view both before the crash and when the "red lines" show up.

ljkenny, 2021-10-02:
The Debian version is bespoke, but fully up-to-date - I can't tell you anymore than that.

The RTL-TCP server is in the loft just above me.  Both the RPi and the Client (Laptop) are connected to the same Wireless LAN.

Sample rate is currently 1.025MHz, but I have tried others - same thing happens.

The terminal does not show the crash logs.  They just stop, hence why I was looking for a debug mode.

Screenshot (even in real life the spectrum is frozen right from connection, as though only one sample is taken):

[image]

Logs:
```
$ sdrpp                                                                                                                                                                                                                                                                                                                                                                                       
[2021-10-02 21:04:05.627] [info] SDR++ v1.0.3                                                                                                                                                                                                                                                                                                                                                 
[2021-10-02 21:04:05.627] [info] Loading config                                                                                                                                                                                                                                                                                                                                               
[2021-10-02 21:04:05.857] [info] Loading icons                                                                                                                                                                                                                                                                                                                                                
[2021-10-02 21:04:05.866] [info] Loading band plans                                                                                                                                                                                                                                                                                                                                           
[2021-10-02 21:04:05.887] [info] Loading band plans color table                                                                                                                                                                                                                                                                                                                               
[2021-10-02 21:04:05.925] [info] Loading modules                                                                                                                                                                                                                                                                                                                                              
[2021-10-02 21:04:05.926] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so                                                                                                                                                                                                                                                                                                        
[2021-10-02 21:04:05.977] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so                                                                                                                                                                                                                                                                                                              
[2021-10-02 21:04:05.994] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so                                                                                                                                                                                                                                                                                                               
[2021-10-02 21:04:06.028] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so                                                                                                                                                                                                                                                                                                            
[2021-10-02 21:04:06.063] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so                                                                                                                                                                                                                                                                                                               
[2021-10-02 21:04:06.091] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so                                                                                                                                                                                                                                                                                                [... cut ...]                                                                                                                
<Lee: About to change frequency>                                                                                                                                                                                                                                                                                                                                                              
$
```

ljkenny, 2021-10-02:
I should also mention that CubicSDR connects and functions properly with the same setup.

AlexandreRouma, 2021-10-02:
Can you show the logs from the rtl_tcp server running on the pi?
It honestly looks like that's crashing when setting some options and then SDR++ tried to send it a tune command which crashes because the socket is closed (which is a known bug and I still have to fix)

ljkenny, 2021-10-02:
```
Oct 02 22:09:17 raspberrypi SoapySDRServer[449]: SoapyServerListener::accept([::ffff:192.168.1.3]:43228)
Oct 02 22:09:17 raspberrypi SoapySDRServer[449]: SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
Oct 02 22:09:18 raspberrypi SoapySDRServer[449]: SoapyServerListener::close()
```

AlexandreRouma, 2021-10-02:
That's not RTL-TCP.... that's SoapyServer,
You're supposed to use the SoapySDR source with it...

ljkenny, 2021-10-02:
I was just coming to that same conclusion myself!

It does work quite a lot better when I use `rtl_tcp` from the command line. :)

Does SDR++ work with Soapy?  I see a SoapySDR option, but no way to enter a remote IP address.

AlexandreRouma, 2021-10-02:
the soapy source indeed doesn't let you enter an IP manually, but soapysdr should automatically detect the server on the network and display it as an option

ljkenny, 2021-10-02:
What if the IP address you wish to connect to is on a WAN?

AlexandreRouma, 2021-10-02:
then it won't work

ljkenny, 2021-10-02:
Is this something you'd be prepared support in the future?

Seems to work well in CubicSDR, but I'd prefer to use SDR++.

AlexandreRouma, 2021-10-02:
Not sure, soapysdr has been causing more issues then good. Using RTL-TCP generally gives better results than soapyserver anyway.
Also there's gonna be SDR++ Server at some point.

gvanem, 2021-10-03:
>  They just stop, hence why I was looking for a debug mode.

It should have `-h, -help` command-line options to hint users on the `-s` option.
I've done this patch:
```diff
--- a/core/src/options.cpp 2021-09-30 11:29:24
+++ b/core/src/options.cpp 2021-10-01 17:29:17
@@ -5,6 +5,16 @@
 namespace options {
     CMDLineOptions opts;

+    void show_help (const char *who_am_I)
+    {
+      printf ("Usage: %s [options]\n"
+              "options:\n"
+              "  -r, --root <root>:  specify root\n"
+              "  -s, --show-console: show console messages\n"
+              "  -S, --server:       enable server-mode\n"
+              "  -h, --help:         show this help\n", who_am_I);
+    }
+
     void loadDefaults() {
 #ifdef _WIN32
         opts.root = ".";
@@ -25,9 +35,13 @@
             else if (!strcmp(arg, "-s") || !strcmp(arg, "--show-console")) {
                 opts.showConsole = true;
             }
-            else if (!strcmp(arg, "--server")) {
+            else if (!strcmp(arg, "-S") || !strcmp(arg, "--server")) {
                 opts.serverMode = true;
             }
+            else if (!strcmp(arg, "-h") || !strcmp(arg, "--help")) {
+                show_help (argv[0]);
+                return false;
+            }
             else {
                 spdlog::error("Invalid command line option: {0}", arg);
                 return false;
```

ljkenny, 2021-10-03:
Yes, it should.

Apart from the odd parameter nomenclature, patch LGTM.

closed 2022-09-29 by AlexandreRouma