AlexandreRouma/SDRPlusPlus #412: Zadig Requirements are not in readme.md

ITCMD, 2021-10-06:
It appears installing an rtl-sdr in sdr++ requires adding a usb driver with zadig. However, users are not told this in the readme for installation.

ITCMD, 2021-10-07:
I do understand your point, however a quick comment even linking to that page saying it is required or a comment saying that the RTL-SDR needs to first be set up would go a long way. 

All of RTL-SDR's setup guides are for using SDR#, which makes it seem like it is specific to SDR# compared to SDR++. Any clarification would be extremely helpful.

Ta-2om, 2022-03-02:
I started SDRpp with all Windows versions. After switching to Linux many months ago, never thought of it..that I have to dealwith Zadig. 

Remembering Zadig installs **Win**USB drivers. Hence we are using Linux, why the need to Zadig is not required on Linux...

Bye

closed 2022-09-29 by AlexandreRouma