AlexandreRouma/SDRPlusPlus #430: Dead lock, when switching one VFO sink from "New audio" to "Null" in a multi-VFO configuration

thomasgi1, 2021-10-10:
Setup:
- running self-compiled SDR++ commit 8deb018 on MacBook Air (13-inch, 2017) macOS 11.6 (Big Sur)
- source is RTL-SDR
- multi-VFO with 3 radios "Radio 1", "Radio 2" and "Radio 3" (all not muted)
- each radio has the "New Audio" sink set
- SDR++ is playing

Now try to change the VFO sink of "Radio 2" from "New Audio" to "Null". SDR++ immediately hangs for ever (spinning beach ball).

Here are the related stack traces:
```
(lldb) thread list
Process 32032 stopped
* thread #1: tid = 0x27b9a7, 0x00007fff2041c4ca libsystem_kernel.dylib`__psynch_mutexwait + 10, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  thread #2: tid = 0x27b9c4, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #5: tid = 0x27b9da, 0x00007fff2041a2ba libsystem_kernel.dylib`mach_msg_trap + 10, name = 'com.apple.NSEventThread'
  thread #13: tid = 0x27b9eb, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #14: tid = 0x27b9ef, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #15: tid = 0x27b9f1, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #16: tid = 0x27b9f4, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #17: tid = 0x27b9f6, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #18: tid = 0x27b9f8, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #19: tid = 0x27b9fa, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #20: tid = 0x27b9fd, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #21: tid = 0x27ba01, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #22: tid = 0x27ba04, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #23: tid = 0x27ba06, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #24: tid = 0x27ba08, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #25: tid = 0x27ba0b, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #26: tid = 0x27ba0d, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #27: tid = 0x27ba10, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #28: tid = 0x27ba14, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #29: tid = 0x27ba17, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #30: tid = 0x27ba1c, 0x00007fff2041a2ba libsystem_kernel.dylib`mach_msg_trap + 10, name = 'org.libusb.device-hotplug'
  thread #31: tid = 0x27ba1d, 0x00007fff2041a2f6 libsystem_kernel.dylib`semaphore_wait_trap + 10, name = 'AMCP Logging Spool'
  thread #37: tid = 0x27ba2d, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #38: tid = 0x27ba2e, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #39: tid = 0x27ba2f, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #40: tid = 0x27ba31, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #41: tid = 0x27ba33, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #42: tid = 0x27ba36, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10, name = 'com.apple.audio.IOThread.client'
  thread #43: tid = 0x27ba4a, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #44: tid = 0x27ba4b, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #45: tid = 0x27ba4c, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #46: tid = 0x27ba4e, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #47: tid = 0x27ba4f, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #48: tid = 0x27ba50, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #49: tid = 0x27ba5e, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #50: tid = 0x27ba5f, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #51: tid = 0x27ba60, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #52: tid = 0x27ba62, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #53: tid = 0x27ba63, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #54: tid = 0x27ba64, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #55: tid = 0x27ba68, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #56: tid = 0x27ba69, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #57: tid = 0x27ba6d, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #58: tid = 0x27ba77, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #59: tid = 0x27ba78, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #60: tid = 0x27ba7a, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #61: tid = 0x27ba7b, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #62: tid = 0x27ba7c, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #63: tid = 0x27ba7d, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #64: tid = 0x27ba7e, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #65: tid = 0x27ba7f, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #66: tid = 0x27ba80, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #67: tid = 0x27ba85, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #68: tid = 0x27ba86, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #69: tid = 0x27ba8e, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #73: tid = 0x27bb18, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #74: tid = 0x27bb19, 0x00007fff2041ccde libsystem_kernel.dylib`__psynch_cvwait + 10
[... 61 lines cut ...]
    frame #3: 0x00007fff22190fc2 CoreAudio`HALB_Mutex::Lock() + 92
    frame #4: 0x00007fff21dcc902 CoreAudio`HALC_ProxyIOContext::StopIOProc(int (*)(unsigned int, AudioTimeStamp const*, AudioBufferList const*, AudioTimeStamp const*, AudioBufferList*, AudioTimeStamp const*, void*)) + 46
    frame #5: 0x00007fff21d98927 CoreAudio`HAL_HardwarePlugIn_DeviceDestroyIOProcID(AudioHardwarePlugInInterface**, unsigned int, int (*)(unsigned int, AudioTimeStamp const*, AudioBufferList const*, AudioTimeStamp const*, AudioBufferList*, AudioTimeStamp const*, void*)) + 390
    frame #6: 0x00007fff221899c8 CoreAudio`HALDevice::DestroyIOProcID(int (*)(unsigned int, AudioTimeStamp const*, AudioBufferList const*, AudioTimeStamp const*, AudioBufferList*, AudioTimeStamp const*, void*)) + 116
    frame #7: 0x00007fff21e75c7b CoreAudio`AudioDeviceDestroyIOProcID + 278
    frame #8: 0x0000000164986ea1 CoreAudio`___lldb_unnamed_symbol18$$CoreAudio + 2663
    frame #9: 0x00000001649943ad CoreAudio`___lldb_unnamed_symbol209$$CoreAudio + 201
    frame #10: 0x0000000164a91bd2 CoreAudio`___lldb_unnamed_symbol3344$$CoreAudio + 30
    frame #11: 0x00007fff21897460 AudioToolboxCore`APComponentInstance::disposeInstance() + 40
    frame #12: 0x00007fff2199cd33 AudioToolboxCore`AudioComponentInstanceDispose + 40
    frame #13: 0x000000012f3f2492 libportaudio.2.dylib`CloseStream + 234
    frame #14: 0x000000012f3436ae new_portaudio_sink.dylib`AudioSink::stop(this=0x0000000109103200) at main.cpp:134:9 [opt]
    frame #15: 0x00000001002c19a7 libsdrpp_core.dylib`SinkManager::setStreamSink(this=0x00000001006f5c48, name="Radio 2", providerName="None") at sink.cpp:230:23 [opt]
    frame #16: 0x00000001002c405a libsdrpp_core.dylib`SinkManager::showMenu(this=<unavailable>) at sink.cpp:360:13 [opt]
    frame #17: 0x000000010020482d libsdrpp_core.dylib`Menu::draw(this=0x00000001006f27b8, updateStates=false) at menu.cpp:118:13 [opt]
    frame #18: 0x00000001001dbf2a libsdrpp_core.dylib`MainWindow::draw(this=0x0000000100321810) at main_window.cpp:486:23 [opt]
    frame #19: 0x00000001001969ad libsdrpp_core.dylib`sdrpp_main(argc=<unavailable>, argv=<unavailable>) at core.cpp:486:29 [opt]
    frame #20: 0x00007fff2046af3d libdyld.dylib`start + 1
(lldb) 
```

The problem is harder to reproduce with new port audio sink. With rtaudio sink it is 100% reproducible with similar static traces.

[... cut ...] conditions (dataReady and readerStop) are false forever. So, thread 1 and thread 42 are in a deadlock situation.
Normally in a single-VFO configuration the readerStop variable would prevent from such a deadlock, because it would be set to true for that VFO. But in a multi-VFO configuration there are multiple streams each having its own readerStop. Only one of them is set to true (that one whose sink is being changed). The other VFOs still wait for data, but obviously dataReady is stuck on false.

thomasgi1, 2021-10-10:
I think the problem is caused by an improper design of the port audio stream callback function. Here is an excerpt from the [port audio documentation](http://www.portaudio.com/docs/v19-doxydocs/writing_a_callback.html):

_Before we begin, it's important to realize that the callback is a delicate place. This is because some systems perform the callback in a special thread, or interrupt handler, and it is rarely treated the same as the rest of your code. For most modern systems, you won't be able to cause crashes by making disallowed calls in the callback, but if you want your code to produce glitch-free audio, you will have to make sure you avoid function calls that may take an unbounded amount of time to execute. **Exactly what these are depend on your platform but almost certainly include the following: memory allocation/deallocation, I/O (including file I/O as well as console I/O, such as printf()), context switching (such as exec() or yield()), mutex operations, or anything else that might rely on the OS.** If you think short critical sections are safe please go read about priority inversion. Windows amd Mac OS schedulers have no real-time safe priority inversion prevention. Other platforms require special mutex flags. In addition, it is not safe to call any PortAudio API functions in the callback except as explicitly permitted in the documentation._

I think the same statement is true for the RTAudio system. 

The current implementation calls dsp::stream::read(), which blocks on a mutex surrounding a conditional wait variable. According to the port audio documentation this is not allowed. Instead the callback function must NEVER block. If not enough data is available for fulfilling the callbacks' request, then some dummy data (usually just silence) must be generated. Similarly if the stream produces too much data than requested by the callback, then the superfluous data must be dropped. As far I understand the SDR++ design the data flow is driven by the source (e.g. RTL-SDR), which produces raw data at a specific rate (e.g.  2.4 Msamples/s). That data is processed and its rate is reduced according to the attached sink (e.g. audio sink @ 48000 samples/s). However, the 48000 samples/s requested by the audio system are measured in the time domain of the audio hardware. In contrast to that the source data rate is measured in the time domain of the RTL-SDR hardware in this example. Since the hardware clocks are never perfect a jitter buffer is needed somewhere in the system, preferably at a place, where inserting or dropping some samples has the least effect on the overall data processing. IMHO that place must be the audio sink module, because it only affects the heard sound, but not a parallel network sink, which forwards data to anther processing software. So, a non-blocking jitter buffer in the audio sink would solve both the data rate matching problem and the deadlock situation.

AlexandreRouma, 2022-09-29:
likely fixed in recent versions (nightlies), reopen if not

closed 2022-09-29 by AlexandreRouma