AlexandreRouma/SDRPlusPlus #483: SDR++ building on ubuntu 21.10 Impish

nmaster2042, 2021-10-30:
I spent time to upgrade my main computer with last ubuntu 21.10.

```
Now I get this when trying to build sdrpp:
[ 77%] Linking CXX executable sdrpp
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_program_take_code »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_once_mutex_unlock »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « GetX86Info »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_program_new_from_static_bytecode »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_program_compile »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_program_set_backup_function »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_program_free »
/usr/bin/ld : core/libsdrpp_core.so : référence indéfinie vers « orc_once_mutex_lock »
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sdrpp.dir/build.make:108 : sdrpp] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:775 : CMakeFiles/sdrpp.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
```
It seems I have all dependencies installed.

Thanks for help.

AlexandreRouma, 2021-10-30:
you need to clean build (aka completely remove the build directory and start over)

nmaster2042, 2021-10-31:
I removed all the SDRPlusplus directory and made a fresh new git clone from repo.
(It erased the build directory inside too)

Same error.

nmaster2042, 2021-11-03:
Don't understand what happens here, do you have an idea ?

AlexandreRouma, 2021-11-03:
no idea.
All I know is Impish seems to be breaking a lot of software.
I don't really have the time to investigate at the moment though

nmaster2042, 2021-11-06:
While building, I tried make VERBOSE=1 to see what command is actually failing.

It's this one:
`
/usr/bin/c++ CMakeFiles/sdrpp.dir/src/main.cpp.o -o sdrpp  -Wl,-rpath,/home/nicolas/Applis/SDR/SDRPlusPlus/build/core: core/libsdrpp_core.so core/libcorrect/lib/libcorrect.a /usr/lib/x86_64-linux-gnu/libOpenGL.so /usr/lib/x86_64-linux-gnu/libGLX.so /usr/lib/x86_64-linux-gnu/libGLU.so -lGLEW -lGLU -lGL -lfftw3f -lglfw -lvolk -lstdc++fs `

I see you have added Impish in automated builds.
Did you get such issue ?

nmaster2042, 2021-11-06:
I'm not an expert in building app from source, I tried to replace the command line from my previous message by this one:

`/usr/bin/c++ CMakeFiles/sdrpp.dir/src/main.cpp.o -o sdrpp  -Wl,-rpath,/home/nicolas/Applis/SDR/SDRPlusPlus/build/core: core/libsdrpp_core.so core/libcorrect/lib/libcorrect.a /usr/lib/x86_64-linux-gnu/libOpenGL.so /usr/lib/x86_64-linux-gnu/libGLX.so /usr/lib/x86_64-linux-gnu/libvolk.so.2.4 /usr/lib/x86_64-linux-gnu/libGLU.so -lGLEW -lGLU -lGL -lfftw3f -lglfw -lvolk -lstdc++fs`

It actually links

Then I can run make again to let all other modules to build.

I found this using this command to find symbols in so files:

`$ nm -C -A *.so 2> /dev/null | grep orc_program_take_code`

All missing symbols where in libvolk.so.2.4, so I added `/usr/lib/x86_64-linux-gnu/libvolk.so.2.4` to the command line.

I don't know why this fails but with this trick I can build sdrpp.
Let me know if you find something about this.

AlexandreRouma, 2021-11-06:
I don't have that issue, you can check the CI runs in the Actions tab and the actual commands used in .github/workflows/build_all.yml

it's likely your environment being out of date or broken

nmaster2042, 2021-11-06:
OK, sorry, I just found the issue on my environment.

In my /usr/lib/x86_64-linux-gnu/ folder I had a libvolk.so.2.4 but no libvolk.so symbolic link, contrary to all the other libs.
So I added it and voilà ! All is fine now.

closed 2021-11-06 by nmaster2042