AlexandreRouma/SDRPlusPlus #484: Wish: a method to start the SDR, possibly via rigctl

LeaseOnLife, 2021-10-30:
On linux here, I am receiving weatherfaxes via an airspyhf. Propagation changes during the day and I can control the receive frequency by sending a netcat command to sdrpp (currently I use crontab for that). 

I would like to be able to turn on/turn off the sdr via remote control. rigctl doesn't seem to have a command reserved for that. gqrx uses the command "U DSP 1" for that. Could something similar be implemented in sdrpp?

AlexandreRouma, 2021-11-05:
Added `\start` and `\stop` commands to the rigctl server

LeaseOnLife, 2021-11-05:
Gracias! I can test with the next amd64.deb release, hopefully someone else can test quicker with a recent pull from git

AlexandreRouma, 2021-11-05:
You can always get the latest commit's binaries at the Actions tab (or if it's easier https://www.sdrpp.org/nightly , It'll redirect you to the latest successful build)

LeaseOnLife, 2021-11-05:
tested with the latest artifact sdrpp, works!

closed 2021-11-05 by LeaseOnLife