AlexandreRouma/SDRPlusPlus #486: Help, please, what about multi-vfo like in sdr#?

santus118, 2021-10-31:
Is SDR++ have this function, to listen many freqs together?

starvald, 2021-10-31:
in the module manager, select radio, add a name for it then click the + icon.
you can add as many as you wish, you'll also need to assign where to send its audio in the sinks module.

santus118, 2021-10-31:
Thanks, it works! But how I can modify freqs of additional "radio"? Or it must be only deleted, then choose freq and add new "radio"?

starvald, 2021-10-31:
you can either click on the radio in the FFT window to select it or 'page up'/'page down' on the keyboard to cycle through them.
once selected you can click somewhere on the FFT window to move it, use mouse wheel scroll or click a bookmark if you have one.

closed 2021-11-10 by AlexandreRouma

nkaramolegos, 2023-04-24:
Can you share how the VFO is working in term of signal processing? Do you use a NCO? I mean is something like channeling (NCO, filtering, etc)?