AlexandreRouma/SDRPlusPlus #488: Segmentation fault (core dumped)  sdrpp

OZ1BV, 2021-11-01:
Hi and thanks for a nice program. 

My OS is Ubuntu 20.04LTS (fresh installation).

I started using a DVB-TV (Bases on RTL) dongle and it works fine.
Later on I got a HackRF One. 
When connection to the HackRF sdrpp terminates with the message (seen in terminal when starting sdrpp from there):

                         28553 segmentation fault (core dumped) sdrpp

I have also Gqrx installed and it works fine with HackRF one.

Best regards, Brian

AlexandreRouma, 2021-11-01:
Hello, see the troubleshooting section of the readme: https://github.com/AlexandreRouma/SDRPlusPlus#sdr-crashes-when-starting-a-hackrf

OZ1BV, 2021-11-01:
Hi, and thanks
After deleting the file I ran into another issue. The audio is dropping out and this message is repeatably showing up in the terminal:
                        RtApiAlsa::callbackEvent: audio write error, underrun.

Best regards, Brian

OZ1BV, 2021-11-01:
Hi again

Found out that if You shift the audio stream to Network or None and back to Audio again, without stopping the sampling, the underrun stops. If I stop the sampling and start it again the underrun issue returns.

Best regards, Brian

AlexandreRouma, 2021-11-01:
This is a rtaudio bug, see this thread on how to fix it:
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

OZ1BV, 2021-11-02:
Thanks, I did not think about this since other DSR programs did work. Anyway everything works perfect now 😊

closed 2021-11-10 by AlexandreRouma