AlexandreRouma/SDRPlusPlus #489: error while loading shared libraries: libvolk.so.2.3

lu9cbl, 2021-11-01:
Hi, i'm trying to install SDRpp in a fresh install of ubuntu 21.10, and fails with the next error.

sdrpp: error while loading shared libraries: libvolk.so.2.3: cannot open shared object file: No such file or directory

For installing i'm using the info describe in the readme.
sudo apt install libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev
sudo dpkg -i sdrpp_debian_amd64.deb

Thanks in advance.

AlexandreRouma, 2021-11-01:
you need to compile from source, the library versions are different so the precompiiled debs won't work

AlexandreRouma, 2021-11-10:
Ubuntu impish binaries are now available -> https://www.sdrpp.org/nightly

closed 2021-11-10 by AlexandreRouma

AubsUK, 2022-11-27:
> You could try making a symlink named libvolk.so.2.3 to libvolk.so.2.4

Unless there's specific requirement for specific versions of libvolk, my distro (a little bit butchered at the moment, hence finding this thread! - Debian 11 with Kernel 6.0) has:

```
$ ll /usr/lib/x86_64-linux-gnu/libvolk.*
-rw-r--r-- 1 root root 3722752 Dec 27  2020 /usr/lib/x86_64-linux-gnu/libvolk.a
lrwxrwxrwx 1 root root      14 Dec 27  2020 /usr/lib/x86_64-linux-gnu/libvolk.so -> libvolk.so.2.4
-rw-r--r-- 1 root root 2710352 Dec 27  2020 /usr/lib/x86_64-linux-gnu/libvolk.so.2.4
```

So unless SDRPP relies on the specific version, and as long as all other distros also have the symlink, it would be ideal if SDRPP could point to the libvolk.so symlink, and be version agnostic.

I tried the bullsye and sid nightly builds.  The sid build wanted 2.5, creating the symlink to 2.4 worked fine as @shuyuan-liu suggested (but I'm still having another issue I'll work through).

I'm not familiar with how Github compiles from source, but I do see in the [CMakeLists.txt](https://github.com/AlexandreRouma/SDRPlusPlus/blob/72b895fc677af4e47cc31b5ee23f82d49f044c3f/core/CMakeLists.txt#L117), it has /sdr-kit/${ANDROID_ABI}/lib/libvolk.so, so I guess that's the nightly build process changes it to the version installed during the 'build' process.

konung-yaropolk, 2024-03-07:
I had the same problem on Arch until I created a symlink from needed to current installed libvolk2 version. That's not a good solution, so is there an option to link this library statically to avoid this common problem on rolling-release distributives?

snachman, 2025-04-12:
@AubsUK 

> You could try making a symlink named libvolk.so.2.3 to libvolk.so.2.4

Thank you very, very much. I had to symlink from 2.5 to 2.4 and also 2.3 but this worked!

On a personal note I absolutely love linux and using symlinks in this way seems fairly obvious now but I hadn't needed it before. Thank you!