AlexandreRouma/SDRPlusPlus #496: [Bug] Freeze when clicking on bookmark label whose pin is outside the pandapter area

shuyuan-liu, 2021-11-02:
Might be related to #495. When a bookmark label's pin / stem / whatever it's called is outside the pandapter area, in addition to the text on the label disappearing, clicking on the label causes SDR++ to freeze. When the pin is inside the display area, clicking on the label correctly shifts the VFO to that bookmark's frequency, despite the label still having no text.

[image]

DeltaHotel4, 2021-11-02:
The same thing happens to me here (OS X 10.14).

gvanem, 2021-11-03:
On Windows in Release-mode, there is an exception:
```
 Security check failure or stack buffer overrun - code c0000409 
```
and an `int 29h`. No further explanation. 

And building for Debug-mode, I get this only:
```
ucrtbased!issue_debug_notification+0x31:
68ce12c1 b001            mov     al,1
```

and this call-stack:
```
ucrtbased!issue_debug_notification(wchar_t * message = 0x68c437d4 "abort() has been called")+0x31
ucrtbased!__acrt_report_runtime_error(wchar_t * message = 0x68c437d4 "abort() has been called")+0x11
ucrtbased!abort(void)+0x1a
ucrtbased!terminate(void)+0x7d
sdrpp!__scrt_unhandled_exception_filter(struct _EXCEPTION_POINTERS * pointers = 0x032fdbd4)+0x74
KERNELBASE!UnhandledExceptionFilter+0x192
ntdll!__RtlUserThreadStart+0x3ce35
ntdll!_RtlUserThreadStart+0x1b
```
Thanks MicroSoft!

AlexandreRouma, 2021-11-03:
The bug happens because when the frequency bookmark is out of view, tuning to it requires to change the view offset, this requires to lock the waterfall's mutex. However, this happens from insider the input handler function of the frequency manager which is called from the waterfall **while the waterfall's mutex is already locked**. This means the code attempts to lock the mutex a second time and, instead of giving any useful information, windows terminates the process...

AlexandreRouma, 2021-11-03:
Should now be fixed in the latest commit, please confirm

shuyuan-liu, 2021-11-03:
Commit ge6377d4 works fine on Linux here. One thing I noticed is that upgrading to this commit (`git pull`, `make`, `make install` over the last version) sets my FFT resolution to 524288; no problems after manually setting it back to my previous setting. I'll close the issue once someone else confirms it's working for them.

AlexandreRouma, 2021-11-03:
it does that when there's an issue with the config, changed it so that it would use the default instead

closed 2021-11-10 by AlexandreRouma