AlexandreRouma/SDRPlusPlus #499: SDR++ install linuxmint 19.2

opabert, 2021-11-03:
Thank you for SDR++
Try install in linuxmint 19.2  install script  libvolk2-dev  not found and change to libvolk1-dev
sudo apt install libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev

These install releases keep asking for the libvolk2-dev or do i pick the wrong ones?
https://github.com/AlexandreRouma/SDRPlusPlus/releases

sudo dpkg -i sdrpp_debian_amd64.deb
I not able to find.

Any advices welcome
Thank you

AlexandreRouma, 2021-11-03:
you didn't download the right deb. Mint 19.2 is based on ubuntu 18.04 bionic

opabert, 2021-11-03:
Aiiiii my dome mistake.  Thank you its working.  
RTLSDR  T820   or E4000 in middle spectrum screen  see spark.  Airspy 3 MHz span some distorted audio  WFM  radio.

AlexandreRouma, 2021-11-03:
for the rtl thing, enable "IQ correction" in the source menu.
For the seconde I'm gonna need more info.
Considering you're running mint I assume it's likely https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

opabert, 2021-11-03:
Found it, thank you. Audio part i need investigate.  Last.  how setup extra VFO?  
After this, close this topic as solved

AlexandreRouma, 2021-11-03:
Module Manager -> in the bottom drop down select "radio" ->give a name to the new radio in the text field next to the drop down -> click "+"
To remove a vfo, go in the list of modules in the module manager and click the "-" button next to the instance of the VFO you wish to remove.

Alternatively, you can also temporarily disable a VFO by unchecking the checkbox next to the menu name (this will completely disable its dsp so it won't use any CPU)

opabert, 2021-11-03:
Okee thank you  i try  the VFO,  btw i set audio to  19200.    
something else  other topic maby?  is  possible  using  SDR++  as a frontend for the gr-imsi  gr-gsm  gr-dect dec who using GNuradio  as the tuner and you know its very limited / difficult find or tuning signals with GNUradio in some cases.  SDR++ will be  a very nice good tuner for all off these.  I think using one or two localhost to pipe to GNuradio tuner, not sure.

AlexandreRouma, 2021-11-07:
Use the raw setting in the radio, set the sink of the radio to "network" and configure it to send samples over udp or tcp to GRC.
Bandwidth is limited by the maximum sample rate of the network sink

closed 2021-11-07 by AlexandreRouma