AlexandreRouma/SDRPlusPlus #502: Kind request: Full screen + Hide menu at startup + Start RX. Kind of Kiosk mode

DocViper666, 2021-11-04:
Hello,

I am a member of a makerspace (with a ham radio club) and we have a screen with SDR++ running at our window (the actual widow to the street).
Currently we have no reliable way to startup SDR++ and go to fullscreen and hide the menu (left side).

xdotool is detecting the application window but the F11 key (or other keys) does not get passed on.

It would be great for a future version to implement a startup feature (either be commandline parameters) or in the menu to 
- start in fullscreen mode (like hitting F11)
- start rx (using last settings)
- optionally hide menu

If someone has a clever idea please let me know.

Thanks a lot.
73 and best regards
Joe, OE5JFE

corvus-ch, 2021-11-04:
Using a tiling window manager such as i3 could be a solution for full screen but I consider this only to be workaround. I second the idea of SDR++ have more control options via keyboard (vim and i3 user here).

AlexandreRouma, 2021-11-04:
Hello, 
You just need to press F11 for it to go full screen

AlexandreRouma, 2021-11-04:
See this page for all the keyboard controls:
https://github.com/AlexandreRouma/SDRPlusPlus/wiki/UI-Controls

DocViper666, 2021-11-04:
> 
> 
> Hello, You just need to press F11 for it to go full screen EDIT: Also the state of the menu is retained between runs, if you hide it, it'll stay hidden when you start it again

Thank you Alexandre,

I know about the commands but it would be great that after a powerloss (of the Raspberry 4) the system would boot and go automatically back to fullscreen and start RX. 

Thanks

AlexandreRouma, 2021-11-05:
I'll check the code, might have forgotten to save the full screen state.
As for starting the SDR on startup, That's not something I can easily add.
You should be able to just simulate the keybind for it (`End` key)

DocViper666, 2021-11-05:
Thank you. We will try a different window manager. Maybe this will work then with xdotool to send the End key.
Btw it is actually running since two weeks ongoing. So good work on stable operation 👍

AlexandreRouma, 2021-11-05:
Great to know!
By the way, in case you didn't notice the feature, since you're running it as a display only, disabling the radio module (checkbox next to the menu name) completely disables the radio's DSP. This decreases significantly the CPU usage so you might be able to run faster SDRs or use lower power hardware to run it

AlexandreRouma, 2021-11-05:
all the tuning can be controlled through the rigctl interface.
For more complex stuff writing a custom module would likely be a better way.

Working on the fullscreen issue rn, seems like I indeed forgot to back up the state in the config

DocViper666, 2021-11-05:
> rigctl interface

That's what we have in mind. Should cover our needs.
Together with https://github.com/AlexandreRouma/SDRPlusPlus/issues/484
also the startup would be solved.

AlexandreRouma, 2021-11-05:
Alright, as of the latest commit it should now save and restore the fullscreen state and I've added `\start` and `\stop` commands to the rigctl server module.

Btw, I'd love to see how the display looks at the makerspace so feel free to send picture!

DocViper666, 2021-11-05:
Excellent. Thanks a lot. I will update next time I am there.
Here a page with projects at display. /dev/radio is the ham radio club with a lots of SDR enthusiasts 
https://devlol.org/wiki/window

AlexandreRouma, 2021-11-07:
Looks nice!
SInce this seems to have been resolved, I'll close the issue

closed 2021-11-07 by AlexandreRouma