AlexandreRouma/SDRPlusPlus #507: fluttery audio fix 112 has errors

DANKC2STA, 2021-11-06:
Hi-

I have sdr++ up and running,OS:Xubuntu 20.04, I have audio out, but it is fluttery. Following directions in issue #112, but i have eerors - see below

/rtaudio/build$ cmake -DCMAKE_INSTALL_PREFIX:PATH="/usr" -DRTAUDIO_API_JACK=OFF -DRTAUDIO_API_ALSA=OFF ..
-- Compiling with support for: pulse
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PULSESIMPLE_LIB
    linked by target "rtaudio" in directory /home/dan/rtaudio
    linked by target "audioprobe" in directory /home/dan/rtaudio/tests
    linked by target "duplex" in directory /home/dan/rtaudio/tests
    linked by target "testall" in directory /home/dan/rtaudio/tests
    linked by target "teststops" in directory /home/dan/rtaudio/tests
    linked by target "record" in directory /home/dan/rtaudio/tests
    linked by target "apinames" in directory /home/dan/rtaudio/tests
    linked by target "playraw" in directory /home/dan/rtaudio/tests
    linked by target "playsaw" in directory /home/dan/rtaudio/tests
PULSE_LIB
    linked by target "rtaudio" in directory /home/dan/rtaudio
    linked by target "audioprobe" in directory /home/dan/rtaudio/tests
    linked by target "duplex" in directory /home/dan/rtaudio/tests
    linked by target "testall" in directory /home/dan/rtaudio/tests
    linked by target "teststops" in directory /home/dan/rtaudio/tests
    linked by target "record" in directory /home/dan/rtaudio/tests
    linked by target "apinames" in directory /home/dan/rtaudio/tests
    linked by target "playraw" in directory /home/dan/rtaudio/tests
    linked by target "playsaw" in directory /home/dan/rtaudio/tests

-- Configuring incomplete, errors occurred!
See also "/home/dan/rtaudio/build/CMakeFiles/CMakeOutput.log".
See also "/home/dan/rtaudio/build/CMakeFiles/CMakeError.log".

=========================================================

Here is "/home/dan/rtaudio/build/CMakeFiles/CMakeOutput.log".

The system is: Linux - 5.11.0-38-generic - x86_64
Determining if the CXX compiler works passed with the following output:
Change Dir: /home/dan/rtaudio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_bcad9/fast && /usr/bin/make -f CMakeFiles/cmTC_bcad9.dir/build.make CMakeFiles/cmTC_bcad9.dir/build
make[1]: Entering directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_bcad9.dir/testCXXCompiler.cxx.o
/usr/bin/c++     -o CMakeFiles/cmTC_bcad9.dir/testCXXCompiler.cxx.o -c /home/dan/rtaudio/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
Linking CXX executable cmTC_bcad9
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_bcad9.dir/link.txt --verbose=1
/usr/bin/c++       CMakeFiles/cmTC_bcad9.dir/testCXXCompiler.cxx.o  -o cmTC_bcad9 
make[1]: Leaving directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: /home/dan/rtaudio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_7986f/fast && /usr/bin/make -f CMakeFiles/cmTC_7986f.dir/build.make CMakeFiles/cmTC_7986f.dir/build
make[1]: Entering directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_7986f.dir/CMakeCXXCompilerABI.cpp.o
/usr/bin/c++     -o CMakeFiles/cmTC_7986f.dir/CMakeCXXCompilerABI.cpp.o -c /usr/share/cmake-3.16/Modules/CMakeCXXCompilerABI.cpp
Linking CXX executable cmTC_7986f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_7986f.dir/link.txt --verbose=1
/usr/bin/c++       CMakeFiles/cmTC_7986f.dir/CMakeCXXCompilerABI.cpp.o  -o cmTC_7986f 
make[1]: Leaving directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'

Determining if the function gettimeofday exists passed with the following output:
Change Dir: /home/dan/rtaudio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_7ab3e/fast && /usr/bin/make -f CMakeFiles/cmTC_7ab3e.dir/build.make CMakeFiles/cmTC_7ab3e.dir/build
make[1]: Entering directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_7ab3e.dir/CheckFunctionExists.cxx.o
/usr/bin/c++    -DCHECK_FUNCTION_EXISTS=gettimeofday   -o CMakeFiles/cmTC_7ab3e.dir/CheckFunctionExists.cxx.o -c /home/dan/rtaudio/build/CMakeFiles/CheckFunctionExists/CheckFunctionExists.cxx
Linking CXX executable cmTC_7ab3e
[... 309 lines cut ...]
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/dan/rtaudio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d9e02/fast && /usr/bin/make -f CMakeFiles/cmTC_d9e02.dir/build.make CMakeFiles/cmTC_d9e02.dir/build
make[1]: Entering directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_d9e02.dir/CheckFunctionExists.cxx.o
/usr/bin/c++    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_d9e02.dir/CheckFunctionExists.cxx.o -c /home/dan/rtaudio/build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx
Linking CXX executable cmTC_d9e02
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d9e02.dir/link.txt --verbose=1
/usr/bin/c++   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_d9e02.dir/CheckFunctionExists.cxx.o  -o cmTC_d9e02  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_d9e02.dir/build.make:87: cmTC_d9e02] Error 1
make[1]: Leaving directory '/home/dan/rtaudio/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_d9e02/fast] Error 2

any help is appreciated, Thanks Dan KC2STA

AlexandreRouma, 2021-11-06:
pulseaudio is not installed

DANKC2STA, 2021-11-06:
terminal says it is installed, latest version 1:13:99 lubuntu 3.12

DAN

AlexandreRouma, 2021-11-06:
Then I don't know what the issue i.
This is a question more appropriate for the rtaudio team since it has nothing to do with the SDR++ codebase.
I recommend opening an issue in the rtaudio repo instead

closed 2021-11-06 by AlexandreRouma

DANKC2STA, 2021-11-06:
ok thanks for your feedback DAN