AlexandreRouma/SDRPlusPlus #508: Key Funcube pro+

landry76560, 2021-11-07:
Hello,

would it be possible to implement the Funcube Pro+ key in your next update?

AlexandreRouma, 2021-11-07:
Hello,
I do not own this dongle so I cannot implement support for it.

nmaster2042, 2021-11-08:
You can try the SoapySDR Driver https://github.com/pothosware/SoapyFCDPP

AlexandreRouma, 2021-11-08:
hmm, didn't know soapy supports it, good to know

AlexandreRouma, 2022-10-03:
emailed the manufacturer about it, may be able to get one to add support for it

vasagle-gleblu, 2022-11-14:
I am able to get SDRPlusPlus to run using SoapyFCDPP.  However, the audio is "choppy".  When I try CubicSDR with the SoapyFCDPP driver I am able to get it to work when I modify the run-time setting (RF sample rate?).  I cannot find a place in SDRPlusPlus to input such a number when choosing the SDR source.

AlexandreRouma, 2022-11-15:
If the issue is caused by RTaudio: https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112
If not, notify the SoapyFCDPP devs, I don't do support for individual soapy modules.

vasagle-gleblu, 2022-11-18:
The author of the SoapyFCDPP driver said the parameter I'm looking for is called the period:

"This setting adjusts the number of samples requested by the SoapyFCDPP driver on each read of the Linux audio system. Smaller values mean smaller blocks of samples, delivered more often - which can help SDR applications to run more smoothly, as you have noticed.

The default value if not specifically set is sample_rate/4, which means the SDR gets samples delivered 4 times a second, leading to 'choppy' audio"

Where can I change this value in the UI of SDR++?
Thank you.