AlexandreRouma/SDRPlusPlus #516: Trouble with SDR++ v1.0.5

srs4511351, 2021-11-16:
I installed the SDR++ v1.0.5 today on my Raspberry Pi 4.

The first time, it ran, but when I zoomed in around 7 MHz, part of the panadapter and waterfall displays were cut off. The signal that I was receiving in the cut off part was still received.

I looked on github and noticed that there were recent fixed, so i reinstalled it with that code.
When I ran sdrpp, it failed with this error:

[2021-11-15 22:38:49.570] [info] SDR++ v1.0.5
[2021-11-15 22:38:49.571] [warning] Root directory /home/dietpi/Library/Application Support/sdrpp does not exist, creating it
terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: cannot create directory: No such file or directory [/home/dietpi/Library/Application Support/sdrpp]
Aborted

I manually created the directory. Now, when running sdrpp, it displayed the error:

[2021-11-15 22:42:10.991] [info] SDR++ v1.0.5
[2021-11-15 22:42:10.991] [info] Loading config
[2021-11-15 22:42:10.991] [warning] Config file '/home/dietpi/Library/Application Support/sdrpp/config.json' does not exist, creating it
[2021-11-15 22:42:10.992] [warning] ConfigManager locked, waiting...
[2021-11-15 22:42:10.992] [error] Resource directory doesn't exist! Please make sure that you've configured it correctly in config.json (check readme for details)

It did create /home/dietpi/Library/Application Support/sdrpp/config.json, but running sdrpp again caused the Resource directory doesn't exist!  error again.

----Steve

AlexandreRouma, 2021-11-16:
Pull the latest version and clean build, there was an option wrongly enabled by default in the latest commit.
You'll also need to reset the configuration by running `rm ~/.config/sdrpp/config.json`

srs4511351, 2021-11-16:
Thanks for the quick action. It runs now.

Now, after recompile, removing settings and running, it still cuts off the panadapter and waterfall when I zoom in as mentioned in my first post.
The more I zoom in, the more is cut off.
It still receives the signal whose display was cut off and zooming out will again display that frequency.

While you were fixing the code, I installed an earlier version and it worked correctly.

----Steve

srs4511351, 2021-11-16:
The version that works for me is v1.0.4 built Oct. 16, 2021

AlexandreRouma, 2021-11-16:
I'm not sure I understand what your issue is (a screenshot of video would help), but I think your issue is the zoom doesn't behave like you expect it to.
If you have a VFO enabled, the zoom with zoom **while trying to keep the VFO in the center of the visible area**.
This means if you want to zoom in on a signal, move your VFO to it and then zoom.

srs4511351, 2021-11-16:
Basically, just tune a signal in the center of the display.
Click the bullseye button (is it the VFO centering button?)
There are 3 screen shots.
1. Completely zoomed out.
2. Zoomed out with tuned signal still displayed.
3. Zoomed out so that the tuned signal is not displayed, but is still being received.
Again, it works differently with v1.0.4, keeping spectrum across the entire window.

[image]
[image]
[image]

AlexandreRouma, 2021-11-16:
Ok that's a bug what the hell, I'll try to reproduce it
Likely one of the changes to the waterfall code I made this weekend.

AlexandreRouma, 2021-11-16:
Ok, can reproduce and I know where the issue is, will update when it's fixed

AlexandreRouma, 2021-11-16:
Should be fix, can you confirm?

srs4511351, 2021-11-16:
Yes, it is working with a few minutes of use.
I think you fixed it, thanks!

----Steve

closed 2021-11-16 by AlexandreRouma