AlexandreRouma/SDRPlusPlus #517: sdrplay not recognised

G1DLC, 2021-11-17:


G1DLC, 2021-11-17:
sdrplay not recognised on boot up?

AlexandreRouma, 2021-11-17:
You haven't given much information so it's hard to say what your issue is.
What operating system are you using?
What SDR++ version?
Which SDRplay SDR?
I suspect the SDRplay service isn't started. On some windows installs and some linux distros, the sdrplay service isn't configured to start at boot, in which case SDR++ won't load any SDRplay device.

G1DLC, 2021-11-17:
operating system windows 11
SDR++ V1.0.4
SDRplay  rspduo
sorry but how do i start sdrplay service ?
dont get old 
John

G1DLC, 2021-11-17:
hi
i have found that if i boot sdruno first then close sdruno sdr++ will work  .
can this be made so sdr++ boots without having to boot sdruno up and then close?

AlexandreRouma, 2021-11-17:
that's because sdruno forces the API to start,
You need to enable starting the API on boot:
1) Open the task manager
2) Go in the "Services" menu
3) Click on "Open Services" at the bottom of the window
4) Scroll down and search for "SDRplay API Service"
5) Right-click "SDRplay API Service" and select "Properties"
6) Change "Startup type" to "Automatic"
7) Click "Apply"
8) Reboot

And it should work like you expect

G1DLC, 2021-11-17:
hi
thanks for your promt replys  to answer the problem 
its working fine now 
dont get old 
john g1dlc

closed 2021-11-24 by AlexandreRouma