AlexandreRouma/SDRPlusPlus #518: Hash collision!!! Fatal error!! - Pop!_OS 21.04

cdehaan82, 2021-11-18:
I have removed all of the SoapySDR modules and installed only what is listed in the notes on here **libsoapysdr-dev** 

[2021-11-18 17:28:55.188] [info] SDR++ v1.0.4
[2021-11-18 17:28:55.188] [info] Loading config
[2021-11-18 17:28:55.364] [info] Using OpenGL 3.0
[2021-11-18 17:28:55.527] [info] Loading icons
[2021-11-18 17:28:55.533] [info] Loading band plans
[2021-11-18 17:28:55.545] [info] Loading band plans color table
[2021-11-18 17:28:55.567] [info] Loading modules
[2021-11-18 17:28:55.567] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2021-11-18 17:28:55.580] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2021-11-18 17:28:55.596] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2021-11-18 17:28:55.615] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2021-11-18 17:28:55.630] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2021-11-18 17:28:55.646] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2021-11-18 17:28:55.665] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2021-11-18 17:28:55.665] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2021-11-18 17:28:55.680] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2021-11-18 17:28:55.697] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2021-11-18 17:28:55.714] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2021-11-18 17:28:55.730] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2021-11-18 17:28:55.746] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2021-11-18 17:28:55.763] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2021-11-18 17:28:55.780] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2021-11-18 17:28:55.796] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2021-11-18 17:28:55.817] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2021-11-18 17:28:55.830] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2021-11-18 17:28:55.847] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2021-11-18 17:28:55.863] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2021-11-18 17:28:55.880] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2021-11-18 17:28:55.897] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2021-11-18 17:28:55.914] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2021-11-18 17:28:55.933] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2021-11-18 17:28:55.947] [info] Initializing Airspy Source (airspy_source)
[2021-11-18 17:28:55.973] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-11-18 17:28:55.989] [info] Initializing Audio Sink (audio_sink)
[2021-11-18 17:28:55.996] [info] Initializing BladeRF Source (bladerf_source)
[2021-11-18 17:28:56.021] [error] Could not list devices -7
[2021-11-18 17:28:56.021] [info] Initializing File Source (file_source)
[2021-11-18 17:28:56.029] [info] Initializing Frequency Manager (frequency_manager)
[2021-11-18 17:28:56.046] [info] Initializing HackRF Source (hackrf_source)
[2021-11-18 17:28:56.068] [info] Initializing LimeSDR Source (limesdr_source)
[2021-11-18 17:28:56.090] [info] Initializing Network Sink (network_sink)
[2021-11-18 17:28:56.096] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-11-18 17:28:56.113] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[2021-11-18 17:28:56.714] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-11-18 17:28:56.715] [info] Initializing Radio (radio)
[2021-11-18 17:28:56.836] [info] RtAudio stream open
[2021-11-18 17:28:56.836] [info] Initializing Recorder (recorder)
[2021-11-18 17:28:56.851] [info] Initializing Rigctl Server (rigctl_server)
[2021-11-18 17:28:56.863] [info] Initializing SDRplay Source (sdrplay_source)
[2021-11-18 17:28:56.879] [error] Module 'sdrplay_source' doesn't exist
[2021-11-18 17:28:56.879] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4build1
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Hash collision!!! Fatal error!!

AlexandreRouma, 2021-11-18:
See [the troubleshooting section of the readme](https://github.com/AlexandreRouma/SDRPlusPlus#hash-collision-error-when-starting)

closed 2021-11-18 by AlexandreRouma

cdehaan82, 2021-11-18:
> See [the troubleshooting section of the readme](https://github.com/AlexandreRouma/SDRPlusPlus#hash-collision-error-when-starting)

I have tried this already as I have said. the only SoapySDR install I did was the libsoapysdr-dev that was in the instructions

AlexandreRouma, 2021-11-19:
You clearly haven't. This error can only show up if the soapy drivers are still installed. 
Run `sudo apt list --installed | grep soapy` and you'll see there are still some device modules left.
The core itself doesn't matter and can stay.

cdehaan82, 2021-11-19:
> You clearly haven't. This error can only show up if the soapy drivers are still installed. Run `sudo apt list --installed | grep soapy` and you'll see there are still some device modules left. The core itself doesn't matter and can stay.

I went back and removed all the SoapySDR links and all that changed is the end of the log is slightly different 
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4build1
Hash collision!!! Fatal error!!

AlexandreRouma, 2021-11-19:
should the output of `sudo apt list --installed | grep soap`
And also show the output of `SoapySDRUtil --find`

paulschow, 2021-12-01:
For anyone else getting this on Pop!_OS 21.04, I had to remove `soapysdr0.7-module-audio`