AlexandreRouma/SDRPlusPlus #522: Can't get pulseaudio on 2nd, nearly identical, machine.

clytle374, 2021-11-23:
Building from current git source.  Both machines are running very similar gentoo builds, both have pulseaudio installed.  Using cmake-gui I get different options on the 2 machines.  Machine #2 shows no pulseaudio configs.
[image]

I'm getting bad audio glitches on machine #2 Any ideas? 
Thanks
Cory

AlexandreRouma, 2021-11-23:
RtAudio bug:
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

clytle374, 2021-11-24:
I think you provided me with correct direction.  While the machines both had pulseaudio installed.  I did not have a global use flag set for pulseaudio set on the 2nd machine.  So media-libs/rtaudio-5.1.0:0  wasn't built with pulseaudio turned on.

 I'm going to close this as the pulseaudio use flag pulled in about 6 hours of rebuilds on that poor old machine, so I'll test it in the morning, but assuming that'll fix it.
Thanks

closed 2021-11-24 by clytle374