AlexandreRouma/SDRPlusPlus #525: Install on MAC HighSierra 10.13.6 fails - fatal error: 'filesystem' file not found = please help

radonclers, 2021-11-26:
### Discussed in https://github.com/AlexandreRouma/SDRPlusPlus/discussions/511

<div type='discussions-op-text'>
Hi All,
I'm not a programmer so please tolerate my ignorance.  After looking at a lot of feedback on your discussions I was very happy to find the article with a download link for a version of SDR++ that will run on an older MacOS.

I have downloaded the precompiled MacOS HighSierra SDR++ 1.05 BUNDLE and installed it on my MacMini running HighSierra.  It did not launch by double clicking on the SDR++ icon.  I was able to get SDR++ to run after I copied (from the downloaded he App package) the plugins folder to: usr/local/lib/sdrpp.  I also copied the folders bandplans, colormaps, fonts, icons and themes to: usr/local/lib/sdrpp.  To make it run I launched the binary sdrpp inside the App package (directory MacOS) by double clicking on it. It launched the terminal app which loaded the binary and executed.
 -->  So it looks like the compiled version is still pointing to the usr directory instead of the internal App package contents.  It would be great to see this fixed.

When I did launch SDR++ everything appeared to work properly with my AirSpy HF+ but **no audio**.  There is no **Audio sink** file (audio_sink.dylib) present in the plugins directory.  Where can I get this ? .... or does the whole package need to be recompiled ?

I tried using the audio_sink.dylib from the Version 1.0.4 package but it didn't load when I launched SDR++.  I suspect this is incompatible or needs to be entered with id in the _CodeSignature/codeResources file inside the compiled MacOS App

Any help would be much appreciated.

Leo

-------------------------------------------------
<sup>Originally posted by **aneitzel** November 12, 2021</sup>
Hello Gyus,

A Lot of MACs are today runs with latest HighSierra 10.13.6 (because of CUDA was supported until this). My imac 27" is from late 2013 (3.5Ghz i7, 32GB RAM, 4GB GPU, 3TB). I tried weeks to get SDRpp run on my MAC without success.

I installed all dependencies via MacPorts. This was ok. Now i cloned the GitHub repo to local, then make. Below the complete output. Error sys FILESYSTEM cant be found? I have no clue what can be wrong here, can someone help me please?

-------------------------------------------------

LINK PINNED ===> here you find a precompiled MacOS HighSierra SDR++ 1.05 BUNDLE https://github.com/AlexandreRouma/SDRPlusPlus/files/7600853/sdr%2B%2B_HighSierra.zip

Extract and copy it to Program folder 

-------------------------------------------------

Alphas-iMac:Downloads alpha$ **git clone https://github.com/AlexandreRouma/SDRPlusPlus.git**
Klone nach 'SDRPlusPlus' ...
remote: Enumerating objects: 9060, done.
remote: Counting objects: 100% (2361/2361), done.
remote: Compressing objects: 100% (892/892), done.
remote: Total 9060 (delta 1603), reused 2005 (delta 1403), pack-reused 6699
Empfange Objekte: 100% (9060/9060), 21.22 MiB | 3.92 MiB/s, fertig.
Löse Unterschiede auf: 100% (4831/4831), fertig.
Alphas-iMac:Downloads alpha$ **cd SDRPlusPlus**
Alphas-iMac:SDRPlusPlus alpha$ **mkdir -p build**
Alphas-iMac:SDRPlusPlus alpha$ **cd build**
Alphas-iMac:build alpha$ **cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_LIMESDR_SOURCE=OFF**

-- The C compiler identification is AppleClang 10.0.0.10001044
-- The CXX compiler identification is AppleClang 10.0.0.10001044
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done

CMake Deprecation Warning at core/libcorrect/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Performing Test COMPILER_SUPPORTS_WPEDANTIC
-- Performing Test COMPILER_SUPPORTS_WPEDANTIC - Success
-- Looking for dotprod in FEC
-- Looking for dotprod in FEC - not found
-- Performing Test HAVE_SSE
-- Performing Test HAVE_SSE - Success
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2") 
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework   
-- Checking for module 'glew'
--   Found glew, version 2.2.0
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.10
-- Checking for module 'volk'
--   Found volk, version 2.5
-- Checking for module 'glfw3'
--   Found glfw3, version 3.3.4
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Checking for module 'libairspyhf'
--   Found libairspyhf, version 1.6
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.6
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version v0.6-20210911-d770add4-MacPorts
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.24
-- Checking for module 'SoapySDR'
--   Found SoapySDR, version 0.8.1
-- Checking for module 'libiio'
--   Found libiio, version 0.23
-- Checking for module 'libad9361'
--   Found libad9361, version 0.2
-- Checking for module 'rtaudio'
--   Found rtaudio, version 5.1.0
-- Configuring done

CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   correct
   fec_shim_shared

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/alpha/Downloads/SDRPlusPlus/build
Alphas-iMac:build alpha$ 
Alphas-iMac:build alpha$ make
[  1%] Building C object core/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/polynomial.c.o
[  1%] Building C object core/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/reed-solomon.c.o
[  2%] Building C object core/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/encode.c.o
[  3%] Building C object core/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/decode.c.o
[  3%] Built target correct-reed-solomon
[  4%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/bit.c.o
[  5%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/metric.c.o
[  7%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/history_buffer.c.o
[  7%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/error_buffer.c.o
[  8%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/lookup.c.o
[  9%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/convolutional.c.o
[  9%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/encode.c.o
[ 10%] Building C object core/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/decode.c.o
[ 10%] Built target correct-convolutional
[ 11%] Building C object core/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/lookup.c.o
[ 12%] Building C object core/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/convolutional.c.o
[ 12%] Building C object core/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/encode.c.o
[ 14%] Building C object core/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/decode.c.o
[ 14%] Built target correct-convolutional-sse
[ 14%] Linking C static library lib/libcorrect.a
[ 14%] Built target correct_static
[ 15%] Building CXX object core/CMakeFiles/sdrpp_core.dir/src/config.cpp.o

/Users/alpha/Downloads/SDRPlusPlus/core/src/config.cpp:5:10: fatal error: 'filesystem' file not found
#include <filesystem>
         ^~~~~~~~~~~~
1 error generated.
make[2]: *** [core/CMakeFiles/sdrpp_core.dir/build.make:76: core/CMakeFiles/sdrpp_core.dir/src/config.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:718: core/CMakeFiles/sdrpp_core.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2

Alphas-iMac:build alpha$ 

---------------------------------------------------------

Thanks a lot!

 </div>

optimolch, 2022-07-02:
Hello @AlexandreRouma, @aneitzel and all,

unfortunately, the problem is still not solved. The program does not seem to work for me. (*) Same symptoms here: No Audio.
Double-clicking does not work. Double-clicking the the sdrpp executable inside the app bundle reports a "damaged" application. I dragged the sdrpp executable to the Terminal.app window, pressed Enter there. This time, SDR++ did launch. 
Terminal output shows missing libraries and plugins. I copied the various missing files from various places inside the SDR++.app bundle to their expected places according to the error messages. But one specific file is missing: audio_sink.dylib. Now SDR++.app can start receiving, but no sound. The Sinks menu does not list an audio sink.
Like @radonclers, I tried it with a "stolen" audio_sink.dylib file from the official SDR++ version 1.0.4 installation. It reports not being able to load it. 

Now what? Does anyone have a working SDR++ for macOS before 10.15 (that does not require compiling)? 
Could the SDR++ macOS requirements be relaxed? (Sometimes it's easy.) This would increase the user base. 

Greetings, Jens

(*) I am using macOS 10.14 Mojave here. I cannot use the newer SDR++ versions since they seem to be built for minimum 10.15 target. Similar to André, I can say that many Mac users have frozen their system version at 10.14 because it was the last that can run 32-bit applications. Non-admin user account, not wanting to install system modifications except Apple's. Receiver is an Airspy HF+ Discovery.

bubnikv, 2022-07-02:
Some C++17 features such as std::filesystem are not supported by OSX/clang on older OSX versions. That's why we are using boost::filesystem instead at PrusaSlicer.

std::filesystem is implemented by a shared library, which is being distributed with OSX system and not updated to C++17.

radonclers, 2022-07-02:
I have managed to get SDR++ working on my 3 macs under High Sierra and Mojave.  I have made a package with instructions.  Some key files need to be copied to USR and OPT folders (invisible).  Permissions for the SDR++ app need to be set to R/W for the user.  It looks like the original compile in the installer points to the USR and OPT folders instead of the folders inside the .app folder.  This could be fixed by modifying the source code and recompiling.  Good luc
[SDR++ working package.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/9032620/SDR%2B%2B.working.package.zip)
k

optimolch, 2022-07-02:
Thank you very much, @radonclers. It works now for me. (Well, sometimes audio output seems to overload soon, distorting the sound, but that's a different issue.) I noticed that I needed to keep libfftw3f.3.dylib in /usr/lib, too.

AlexandreRouma, 2022-07-02:
`-DOPT_OVERRIDE_STD_FILESYSTEM=ON` in the cmake command

closed 2022-09-29 by AlexandreRouma