AlexandreRouma/SDRPlusPlus #531: Warbling distorted audio from SDR++ SDRplay RSP2 + Linux Mint 20.2

SheetLightning, 2021-12-03:
I have successfully compiled and installed SDR++ from sources on my Linux Mint 20.2 machine. I amended CMakeLists.txt to enable the SDRplay option. I also turned off Airspy and HackRF and left RTLSDR turned on. I already had the necessary SDRplay  API installed as well as SoapySDR and SoapySDRplay libraries installed. These work fine with CubicSDR.

The compilation ran fine and the program was installed by finally running 'sudo make install'.

The program launches and the GUI is shown. I can choose SDRplay from the list of available sources and my RSP2 is detected and the device number is shown. Whether I receive on AM or FM, the audio is distorted and warbling. Adjusting the bandwidth from 2MHz to 8MHz helps a little but the LNA gain and possibly IF gain has to be adjusted to at least around 70% and even then it does not go away entirely and comes back after reception is paused and re-started.  At these gain levels the waterfall by then looks very orange/yellow and not very useful.

I am not sure where to begin diagnosing this so need a little help.

AlexandreRouma, 2021-12-03:
rtaudio bug:
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

SheetLightning, 2021-12-03:
Thank you. yes following the steps there fixed it. Audio is now clear.

closed 2021-12-03 by AlexandreRouma