AlexandreRouma/SDRPlusPlus #532: No way to enable bias-t for rsp1a running remotely through Soapy

cdbrinker, 2021-12-04:
Connects to remote computer hosting the rsp1a and pipes in the signal just fine but no way to enable bias-t. I can get strong FM stations poorly, the rest of the bands are just white noise. Did I build incorrectly? When running on the rsp++ machine with the SDRPlay source, no problems. CubicSDR has no trouble with Soapy on these two machines and even enables the bias-t by default.

AlexandreRouma, 2021-12-04:
The soapy source doesn't support bias-t remote or not. SoapySDR doesn't have a standard way of enabling bias-t.

AlexandreRouma, 2021-12-04:
Also I'll add that his sounds more like a gain issue, I doubt any SDR would have trouble receiving broadcast FM with its gain set correctly

cdbrinker, 2021-12-04:
> The soapy source doesn't support bias-t remote or not. SoapySDR doesn't have a standard way of enabling bias-t.

Not sure I understand. I can enable bias-t (or FM notch, DAB notch, etc) on CubicSDR via a remote Soapy server. So is there specific support in CubicSDR for my RSP1a that is not present in SDR++?

Just to clarify, the reception in SDR++ via Soapy remote is exactly the same on all bands as with bias-t disabled in CubicSDR via Soapy remote.

AlexandreRouma, 2021-12-04:
I suspect cubic has a list of SDRs and their option name to enable bias-t, then it uses that to enable or disable it. The soapy source in SDR++ is completely device agnostic. Not sure what I can do without manually checking for every device type which would defeat the purpose of the soapy source.

I think I might be able to add a custom device arg field to the soapy source, but you'll have to type in the right soapy args for it to turn on the bias-t

nmaster2042, 2021-12-04:
I don't think Cubic has a list of SDRs, everything is dynamic.

In Soapy driver, there is function to get device options (if avaliable), it's called getSettingInfo()
For SoapySDRPlay3 driver, line 1085 https://github.com/pothosware/SoapySDRPlay3/blob/master/Settings.cpp
But other drivers may have too with other options, depending of the hardware specifications, driver implementation.

And there is a writeSetting() line 1343 and readSetting() line 1533.
The issue I think is to generate device settings dynamically to provide it on the GUI.

Hope this can help

nmaster2042, 2021-12-04:
Checked  SoapyAirspy driver, https://github.com/pothosware/SoapyAirspy/blob/master/Settings.cpp line 363, there is a getSettingInfo() too and writeSetting() line 390 + readSetting() line 408.

So it seems there is a standard way to get device options from a soapy driver.

closed 2022-08-29 by AlexandreRouma