AlexandreRouma/SDRPlusPlus #533: Overlay bookmark labels

ah1102, 2021-12-04:
Bookmark labels overlap.

In the gqrx program, adjacent labels are shifted up and down relative to each other. This avoids overlap.

[image]

AlexandreRouma, 2021-12-04:
This is the intented behavior (same behavior as WebSDR for example)
The way it's done in GQRX quickly clogs up the screen when many bookmarks are used

ah1102, 2021-12-04:
I find this uncomfortable. I would like at least an optional checkbox to enable / disable different types of labels.

shuyuan-liu, 2021-12-04:
Or, at least give the text labels an outline / shadow so overlapping labels don't blend into one big confusing block.

ah1102, 2021-12-06:
Still not enough different colors to highlight different groups of bookmarks

ageras1, 2021-12-22:
AlexandreRouma i can not thank you much about your software.
I think the that gqrx has the best implementation at bookmarks (and color palette, my biased opinion). 
WebSDR looks to me as a bad implementation or something that it works but....
as much as i love gqrx i think sdr++ is going to be the default. but sdr++ is influenced mostly by sdr# (that is nice of course).please consider looking at gqrx and how they implement some things

darauble, 2022-06-03:
Hi @AlexandreRouma ,

I also have "issues" with clogged bookmarks. My usecase where overlapping disturbs is when I have different programs on the same frequency (e.g. radios sharing different frequencies, like BBC or Radio Farda on shortwave).

I don't mind by adding this feature myself, except that I am not a GUI programmer and have no experience with Qt5.

Regarding vertical clogging I'd also add user configurable limit on how many lines can bookmarks use vertically and then start to overlap.

However I don't know, how to check if new rectangle would overlap existing one. I believe that when method to draw filled rectangle is called, it is essentially "lost". Unless there's a method to check for pixel color already drawn?

Other way would be to store coordinates of already drawn rectangles into vector/array and check overlapping mathematically. Checking of this growing list would also increase in time for each subsequent bookmark, but I think this is not a very expensive operation.

Would you please advise on which path to take? I will definitely try it, but your input would be highly appreciated.

darauble, 2022-06-21:
I have tried to play around with code and create a separate Bookmark Manager as a separate plugin on top of the Frequency Manager base. Works for me now quite well:

[image]

If @AlexandreRouma would see this as a worthwhile thing I can share it as a pull request to the main Frequency Manager. Otherwise I can share it as a plugin, but people will need to compile it themselves.

darauble, 2022-06-23:
I have copied Frequency Manager and created my own plugin for exactly this issue:

https://github.com/darauble/bookmark_manager

[image]

betzburger, 2022-09-09:
Um. Just saw this thread. Is this plugin somewhere available? I do like the idea of labels that do not overlap. Especially if you have many entries. 
This brings up a second question: Is is possible to import something like the EIBI list into SDR++?

darauble, 2022-09-09:
As you see right above your comment, repository owner deleted the comment as he does not like someone else to "advertise" their plugins. So as not to add links or "advertisement" I can only say google for "bookmark_manager", maybe you'll find it.

darauble, 2022-09-09:
Regarding EIBI list or something else: no. But it is possible to write a script which would convert it to a JSON file in SDR++'s format.

betzburger, 2022-09-09:
> As you see right above your comment, repository owner deleted the comment as he does not like someone else to "advertise" their plugins. So as not to add links or "advertisement" I can only say google for "bookmark_manager", maybe you'll find it.

Don't know why that is, because it is a great thing if the community adds more features. Anyhow, thank you for your quick reply. I FOUND IT !

AlexandreRouma, 2022-10-10:
The comment was deleted because you posted it in (iirc) 3 different issue. A single one of them was left alone.

closed 2022-10-10 by AlexandreRouma