AlexandreRouma/SDRPlusPlus #537: Remote SoapySDRServer chopping sample input

pauluzs, 2021-12-06:
Hi first thanks for this nice project,

There seem to be a problem when using SDRpp with SoapySDRServer, the sample input is not smooth resulting in hick ups.
Normaly SoapSDRServer can easily stream  2.16Msps from a RTL-SDR and 12Msps from a LimeSDR  (2m/70cm) simultaneous to 2 instances of i.e. cubicSDR.

When trying SoapySDRServer with SDRpp samples coming in are chopping in the spectrum/waterfall  as well  as on the audio output . Even when only using a single device at low rates (the rtl seems worse). 

Altough the problem seems SDRpp related to exclude:

- Tried Soapy 0.7 and 0.8 (no difference, 0.8 show correct samplerate for Lime)
- Tried Wired via Router, Wired Direct and SoapyRemote on localhost (no difference)
- Tried SDRpp Release (SDR++ v1.0.4.deb) as well as Source builds (no difference)

Any help would be appreciated

pauluzs, 2021-12-10:
Adding: tried some different block sizes in the soapy source block (line472), also without success

pauluzs, 2022-01-26:
sdrpp --server solves this issue

closed 2022-01-26 by pauluzs

jerryaaronson, 2025-04-13:
I am using sdrpp --server and have a similar issue. The connection will show anything from 0Mbs to over 90Mbs at times and gets caught in what sounds like loops or maybe more accurately, it sounds like an echo .