AlexandreRouma/SDRPlusPlus #542: Apple M1 native build

grbman, 2021-12-12:
Hi,
I was originally failing to build SDR++ for the M1 using brew (the main problem comes down to [Volk](https://github.com/gnuradio/volk), which relies on [cpu_features](https://github.com/google/cpu_features) where there is a bug that prevents it from compiling on M1. 
However, there is an open pull request for a fix from user [sbehnke and their forked repo of cpu_features](https://github.com/sbehnke/cpu_features) 
After a bit of trial and error, I was able to natively compile SDR++ using the instructions in the readme with a few more steps:

- Clone the Volk repo
- Change the submodule URL for cpu_features to sbehnke's repo, and resend the submodule
- cmake, make, then make install Volk (it should now be able to succeed)
- clone the SDR++ repo, and run cmake _without_ the DOPT_BUILD_M17_DECODER on

Until Google's cpu_features pulls in the fix, the above is a workaround.
I'm not sure how to contribute this fix, since the problem is with a dependency. However, I wanted the info to be out there for others.

grbman, 2021-12-12:
here's the link to the pull request that I mentioned above: https://github.com/google/cpu_features/pull/150

markc1984, 2021-12-20:
Thanks for this info. I've tried compiling Volk by changing the .gitmodules url for cpu_features to https://github.com/sbehnke/cpu_features.git and resyncing, but when I try to compile Volk I get this error message:

[ 11%] Building C object cpu_features/CMakeFiles/cpu_features.dir/src/cpuinfo_arm.c.o
In file included from /Users/**/volk/cpu_features/src/cpuinfo_arm.c:15:
/Users/**/volk/cpu_features/include/cpuinfo_arm.h:118:2: error: "Including cpuinfo_arm.h from a non-arm target."
#error "Including cpuinfo_arm.h from a non-arm target."

Any ideas?

AlexandreRouma, 2022-10-04:
since this just a tip for people facing this problem and is not a SDR++ bug, I'll close the issue. People interested in an actual fix will have to follow the cpu_features project instead.

closed 2022-10-04 by AlexandreRouma