AlexandreRouma/SDRPlusPlus #548: Doesn't start on Windows 11

SergiuEpure, 2021-12-15:
After show Initializining Radio (radio) app crash and dont start

AlexandreRouma, 2022-01-13:
Have gotten many reports of it working fine, I think there's another issue.
The crash on radio load has been reported before so more debugging is needed.
Can you run sdrpp.exe inside a terminal with the `-s` parameter and post what it says in here

iAmJohnBe, 2022-01-13:
As requested:

Suggestion [3,General]: The command sdrpp.exe was not found, but does exist in the current location. Windows PowerShell does not load commands from the current location by default. If you trust this command, instead type: ".\sdrpp.exe". See "get-help about_Command_Precedence" for more details.
PS C:\sdrpp_windows_x64> .\sdrpp.exe -s
[2022-01-13 10:18:38.325] [info] SDR++ v1.0.4
[2022-01-13 10:18:38.326] [info] Loading config
[2022-01-13 10:18:38.484] [info] Using OpenGL 3.0
[2022-01-13 10:18:38.739] [info] Loading icons
[2022-01-13 10:18:38.752] [info] Loading band plans
[2022-01-13 10:18:38.760] [info] Loading band plans color table
[2022-01-13 10:18:38.763] [info] Loading modules
[2022-01-13 10:18:38.764] [info] Loading ./modules/airspyhf_source.dll
[2022-01-13 10:18:38.766] [info] Loading ./modules/airspy_source.dll
[2022-01-13 10:18:38.768] [info] Loading ./modules/audio_sink.dll
[2022-01-13 10:18:38.770] [info] Loading ./modules/bladerf_source.dll
[2022-01-13 10:18:38.773] [info] Loading ./modules/discord_integration.dll
[2022-01-13 10:18:38.774] [info] Loading ./modules/file_source.dll
[2022-01-13 10:18:38.776] [info] Loading ./modules/frequency_manager.dll
[2022-01-13 10:18:38.778] [info] Loading ./modules/hackrf_source.dll
[2022-01-13 10:18:38.780] [info] Loading ./modules/limesdr_source.dll
[2022-01-13 10:18:38.782] [info] Loading ./modules/m17_decoder.dll
[2022-01-13 10:18:38.784] [info] Loading ./modules/meteor_demodulator.dll
[2022-01-13 10:18:38.785] [info] Loading ./modules/network_sink.dll
[2022-01-13 10:18:38.786] [info] Loading ./modules/plutosdr_source.dll
[2022-01-13 10:18:38.789] [info] Loading ./modules/radio.dll
[2022-01-13 10:18:38.791] [info] Loading ./modules/recorder.dll
[2022-01-13 10:18:38.793] [info] Loading ./modules/rigctl_server.dll
[2022-01-13 10:18:38.794] [info] Loading ./modules/rtl_sdr_source.dll
[2022-01-13 10:18:38.796] [info] Loading ./modules/rtl_tcp_source.dll
[2022-01-13 10:18:38.797] [info] Loading ./modules/sdrplay_source.dll
[2022-01-13 10:18:38.800] [info] Loading ./modules/soapy_source.dll
[2022-01-13 10:18:38.801] [error] Couldn't load ./modules/soapy_source.dll. Error code: 126
[2022-01-13 10:18:38.801] [info] Loading ./modules/spyserver_source.dll
[2022-01-13 10:18:38.802] [info] Initializing Airspy Source (airspy_source)
[2022-01-13 10:18:38.818] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-13 10:18:38.841] [info] Initializing Audio Sink (audio_sink)
[2022-01-13 10:18:38.842] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-13 10:18:38.858] [error] Could not list devices -7
[2022-01-13 10:18:38.858] [info] Initializing File Source (file_source)
[2022-01-13 10:18:38.859] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-13 10:18:38.860] [info] Initializing HackRF Source (hackrf_source)
[2022-01-13 10:18:38.874] [info] Initializing LimeSDR Source (limesdr_source)
[2022-01-13 10:18:38.875] [info] Initializing Network Sink (network_sink)
[2022-01-13 10:18:38.877] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-13 10:18:38.878] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2022-01-13 10:18:39.449] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-13 10:18:39.456] [info] Initializing Radio (radio)
[2022-01-13 10:18:39.523] [info] RtAudio stream open
[2022-01-13 10:18:39.524] [info] Initializing Recorder (recorder)
[2022-01-13 10:18:39.526] [info] Initializing Rigctl Server (rigctl_server)
[2022-01-13 10:18:39.528] [info] Initializing SDRplay Source (sdrplay_source)
[2022-01-13 10:18:39.530] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2022-01-13 10:18:39.531] [info] Initializing SoapySDR Source (soapy_source)
[2022-01-13 10:18:39.532] [error] Module 'soapy_source' doesn't exist
[2022-01-13 10:18:39.533] [info] Initializing SpyServer Source (spyserver_source)
[2022-01-13 10:18:39.535] [info] Loading color maps
[2022-01-13 10:18:39.536] [info] Loading ./res/colormaps/classic.json
[2022-01-13 10:18:39.537] [info] Loading ./res/colormaps/classic_green.json
[2022-01-13 10:18:39.539] [info] Loading ./res/colormaps/electric.json
[2022-01-13 10:18:39.540] [info] Loading ./res/colormaps/gqrx.json
[2022-01-13 10:18:39.542] [info] Loading ./res/colormaps/greyscale.json
[2022-01-13 10:18:39.544] [info] Loading ./res/colormaps/inferno.json
[2022-01-13 10:18:39.546] [info] Loading ./res/colormaps/magma.json
[2022-01-13 10:18:39.547] [info] Loading ./res/colormaps/plasma.json
[2022-01-13 10:18:39.549] [info] Loading ./res/colormaps/turbo.json
[2022-01-13 10:18:39.550] [info] Loading ./res/colormaps/viridis.json
[2022-01-13 10:18:39.551] [info] Loading ./res/colormaps/websdr.json
[2022-01-13 10:18:39.566] [info] New DSP samplerate: 2048000.0 (source samplerate is 2048000.0)
[2022-01-13 10:18:39.568] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864
[... 2 lines cut ...]
[2022-01-13 10:18:39.574] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864
[2022-01-13 10:18:39.605] [info] RtAudio stream open
[2022-01-13 10:18:39.621] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864
[2022-01-13 10:18:39.623] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864
[2022-01-13 10:18:39.625] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864
[2022-01-13 10:18:39.628] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 851537500.0!
[... cut ...]
[2022-01-13 10:18:39.631] [info] Running post-init for RTL-TCP Source
[2022-01-13 10:18:39.631] [info] Running post-init for Radio
[2022-01-13 10:18:39.632] [info] Running post-init for Recorder
[2022-01-13 10:18:39.632] [info] Running post-init for Rigctl Server
[2022-01-13 10:18:39.633] [info] Running post-init for SDRplay Source
[2022-01-13 10:18:39.633] [info] Running post-init for SpyServer Source
[2022-01-13 10:18:39.633] [info] Ready.
[2022-01-13 10:18:39.633] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 851537500.0!

AlexandreRouma, 2022-01-13:
interesting, I suspect a config file issue. Is this a clean install?
Could you send a zip with all generated .json files?

iAmJohnBe, 2022-01-13:
yes, I downloaded it from releases lastnight. But I'm not experiencing the problem the op has. my install works great!

AlexandreRouma, 2022-01-13:
ah alright ok I thought you were having issues too.
then I'll need OP to do the same and send back the logs

aosync, 2022-08-17:
Issue has staled, feel free to reopen.

closed 2022-08-17 by aosync