AlexandreRouma/SDRPlusPlus #555: Feature Request - Highlight Active "Radio"

itstheburns, 2021-12-20:
Hello! First off, this is a great piece of software. I have just started getting into SDR so please forgive me if my terms are incorrect. I'm also new to using Github. I keep returning to this software over all of the others I have used. I would like to make a suggestion for some features. I currently have multiple "radios" listening to multiple frequencies, mostly air traffic.

Just like Bookmarks display the name on the FFT, adding the "radio" name as well would make it easier to identify.
Having a better way to select which radio is being tuned and highlight which is active would be great. Currently, you have to click on the tune bar. This can be difficult at times. I have had to realign tunings due to not knowing which is active. 

[image]
[image]

AlexandreRouma, 2021-12-20:
> Just like Bookmarks display the name on the FFT, adding the "radio" name as well would make it easier to identify.

This is to avoid clogging up the FFT with too much text. If you want to know the name of a VFO, hover it and it'll show its name.

> Having a better way to select which radio is being tuned and highlight which is active would be great

The selected radio has a red tuning line while the unselected ones have a yellow line. To select a VFO you either click on it or you can cycle through VFOs using the Page-Up and Page-Down keys on your keyboard.
As for highlighting the menu, I don't think it's a good solution. First it's quite "tacky" i'd say, but also it's completely useless if the menu is scrolled down to somewhere where all radio menus aren't visible.

I'll think of a better way to do this.

itstheburns, 2021-12-21:
> This is to avoid clogging up the FFT with too much text. If you want to know the name of a VFO, hover it and it'll show its name.

I don't feel too much text is an issue, especially on 4k monitors. Having the option to toggle this on and off would be a way to give the user an option to display it or not.

> The selected radio has a red tuning line while the unselected ones have a yellow line.

The selected radio having a red tuning line, this is hard to see sometimes, unless zoomed in. If the active radio menu was highlighted on the left column, or the text changed color, this would be a nice way to see it at a glance. 

> As for highlighting the menu, I don't think it's a good solution. First it's quite "tacky" I'd say, but also it's completely useless if the menu is scrolled down to somewhere where all radio menus aren't visible.

I think its more function over form. I wouldn't consider this "tacky". Finding a nice way to implement function while keeping form is nice. I, personally, keep the menu open so I can change certain settings, which I do frequently, especially when changing  modulation type.

These ideas are just my ideas to make it better. I like the software and its function. It's much more user friendly than others.

closed 2022-09-29 by AlexandreRouma