AlexandreRouma/SDRPlusPlus #556: Recorder behaving weird

zeke800, 2021-12-22:
Hi! Thank you for the great piece of software :) I really enjoy using it for most of my satellite passes. I succeeded in receiving HRPT. Today I stood for 20 minutes out in the snow holding my 1m dish :) I made sure I recorded it, and the recorder had a red "recording 4:06" sign next to it. In the middle of the pass, SDR++ crashed since my computer fell asleep and I restarted it and started recording again. After the pass, I looked in my recordings, and it showed one recording with 4 minutes (the computer crashed at about 11) and another that took up 3.36Gb but the length was 0 seconds... this is the second or third time when this happens. The recorder usually does not record the whole pass, even though it says it does. Please help me on this, as this is a great software and I would like to continue using it :)

shuyuan-liu, 2021-12-23:
The WAV format only supports a maximum file length of 4GB. When this size is exceeded, an incorrect, shorter file length is stored in the WAV header, so any software that reads the length from the header (i.e. most software) will only process a portion of the file. SDR++'s file source ignores the length indicated by the header and plays the whole file regardless, so at least you are able to play back an oversized recording in SDR++.

`sox` has an option to disregard the length info in WAV files and I sometimes use it to cut and stitch oversized recordings into 4GB blocks. You could also convert the recordings from 16-bit samples to 8-bit to cut the size in half, although with some loss in precision.

zeke800, 2021-12-23:
Thank you so much! Any idea why it recorded my previous passes so well? Also - I am on windows - is there any way to check whether or not the 4 Gb limit was reached? How do I continue recording my passes? I don't want to constantly press record over and over :)

shuyuan-liu, 2021-12-23:
I'm not sure why it worked before but now doesn't - maybe your previous recordings happened to be short enough to fit within 4GB, or they were done with a lower sample rate? 20 minutes at a sample rate high enough for HRPT is definitely more than 4GB.

SDR++ currently doesn't tell you the recording's size so you'll have to check it yourself in Windows Explorer / watch the time, and start a new recording every time it gets close to 4GB. You can calculate the maximum recording time per file (that doesn't break the WAV header) by taking 4GB, dividing by 4 bytes (16 bits in each value, 2 values per sample), and dividing that by your sample rate.

The RigCTL Server module in SDR++ allows you to start and stop recording over a TCP connection to it, so in theory you could write a script to automate the process.

You could also use a lower sample rate, as long as it's high enough to contain the HRPT signal.

zeke800, 2021-12-26:
Thanks for the info! I use an RTL-SDR V3 for HRPT, I know this isn't the best but it works for lower-rate HRPT like NOAA. I sample at 2.56MSPS so it is not possible to go lower. Is it possible to send the command you use to split the files? Thanks!

shuyuan-liu, 2021-12-26:
SDR++ records 16-bit samples; each sample consists of two 16-bit integers and is thus 32 bits or 4 bytes long. 4GiB, or 4×1024³ bytes, can contain 1024³ such samples. At 2.56Msps this is 1024³ / (2.560×10⁶) ≈ 419.43s, or just under 7 minutes. As long as each segment you cut is less than 7 minutes it will be a valid WAV file. I will use a segment length of 6 minutes as example.

Say the original recording is `orig.wav` and it's between 8GiB and 12GiB large. The following slices it into three files:
`sox --ignore-length orig.wav seg1.wav trim 0 6:00`
`sox --ignore-length orig.wav seg2.wav trim 6:00 12:00`
`sox --ignore-length orig.wav seg3.wav trim 12:00` (runs to the end of `orig.wav`)

Of course your recording might fit in 2 segments or it might need more than 3.

AlexandreRouma, 2021-12-26:
Instead of turning it into multiple smaller files, I'd recommend converting the file either to raw 16bit data or to wav64.
Cutting it into piece won't allow for correct decoding of an entire image for HRPT.

Note that tools like satdump handle invalid length in the header correctly so you can use it directly with the SDR++ files.

zeke800, 2021-12-27:
I use LeanHRPT, and since I just got my setup working I prefer it over satdump due to the fact that my signal isn't the best (but I get a ~1500 line clear image usually). How do I convert it to 16bit raw or wav64? I tried to load it up in Audacity to convert it to RAW and (obviously) it showed that there were 0 seconds in the file. Thanks :)

closed 2022-09-29 by AlexandreRouma