AlexandreRouma/SDRPlusPlus #557: Need SDRPlusPlus/SDR++ to run on our M1 and other Apple silicon Macs! Help!

NH7L, 2021-12-25:
Lots of us really want SDRPlusPlus SDR++ to run on our M1 and other Apple silicon Macs. Not at all happy being compelled to use CubicSDR, which is a non-intuitive pain in the ass. Most of us non-engineers, non-coders grew up on SDR#. SDR++ is the answer to our dreams. Hope you can help us, Alexandre! Many thanks for all you've done to create this great program.

AlexandreRouma, 2021-12-25:
SDR++ already runs natively on M1 just fine.
I dont own a mac tho so I can only provide intel builds built by the CI
system. (The .app bundle is available on https://www.sdrpp.org/nightly).
I've recently been lent a remote M1 mac mini but i only use it for release,
so the builds arent up to date)

NH7L, 2021-12-25:
Thank you, Alexandre, for answering me, and so quickly.

I realize SDR++ can be made to run on my M1 Mac, but I have no
background or knowledge in software engineering, coding, etc., so that is
beyond me.

I guess my alternative is to install Windows OS on my Mac and then install
the Windows version of your excellent app. I have installed CubicSDR, but I
find it awkward and awful to use. It is totally anti-intuitive. I may have
to learn it, though. Or I could install your program on my Windows laptop.  I
dearly want to run SDR++ on my Mac, however.

Like so many others, I learned what little I know about SDR through using
SDR# on the ancient Windows machine I keep. i am an amateur radio
enthusiast, U.S. license NH7L, and in general, most of us Mac radio
amateurs find ourselves compelled to keep the hated Windows machines around
just for amateur radio purposes, since there is relatively little available
for Macs.

Your app appears to be so good that it would become THE default SDR app for
Macs if it were easier to install. There are thousands and thousands of us!

Best wishes, merry Christmas and happy New Year.

Ernest Murphy NH7L
Birch Bay, Washington state, USA

https://www.qrz.com/db/nh7l

AlexandreRouma, 2021-12-25:
Unfortunately, making the build process on MacOS easier isn't possible, it's just the nature of software development.

And for installing, as I said, main releases have M1 builds when I'm able to have access to a M1 mac. Steve Brightman from SDRplay lent me a remote one, but I can't have it automatically build commits like the CI does, so it requires me to manually build and package the app (which is why I only do it on big releases). Buying a M1 mac is a ridiculously expensive investment (around a year of patreon savings) that I just can't justify.

NH7L, 2021-12-25:
I understand. Apple has always been more problematic for radio apps than
Windows and Unix, I think, though I don't really know enough to say that.

If I can install it easier on my Mac with Windows running, I'll do that.
Windows for Mac isn't as expensive as a new Mac for sure ;) Meanwhile, I'm
going to put your great app on my 2013 Toshiba Satellite, my ham radio-only
machine. SDR# has become bloatware and I've fallen out of love with it.
You've taken the best aspects of SDR# and created something better, whether
I put it on my Mac or not.

Thanks again, Alexandre. Hey: I'm an old guy (73), but from 1951 to 1954,
my family and I (U.S. military/diplomatic people) lived in Brussels and
loved being there. Naturally we visited Leige quite a few times.

Ernest Murphy NH7L

AlexandreRouma, 2021-12-25:
Here's a native M1 build I built today
[SDR++.app.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7776028/SDR%2B%2B.app.zip)

AlexandreRouma, 2021-12-25:
Here is a native M1 build I just made.
(ignore the previous version I posted which was broken)
[sdrpp_macos_m1.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7776043/sdrpp_macos_m1.zip)

NH7L, 2021-12-25:
Alexandre : Many, many thanks!

I just installed sdrpp macos m1.zip on my M1 Macbook Air,  and it started
right up. Have only had time today to test on local FM broadcast, and it
works perfectly.

This build of SDRPP just started up and ran like a clock on a M1 Mac
operated by a software ignoramus. I'd say that's a pretty meaningful beta
test. Have you any plans to announce this to the many Mac-using SDR
enthusiasts out there? I will keep my mouth shut about it ... it's not my
place to do otherwise. But I know your app will be hugely popular among Mac
users.

Thank you again, and happy holidays to you!

Ernest Murphy NH7L
Birch Bay, Washington state, USA

NH7L, 2021-12-25:
A little more information, in case it is useful to you: My Mac is an M1
Macbook Air, USA model, bought new in June 2021. Running MacOS Monterey
Version 12.1 with 8GB memory. From the laptop's own self-description:

*Hardware Overview:*

  Model Name: MacBook Air

  Model Identifier: MacBookAir10,1

  Chip: Apple M1

  Total Number of Cores: 8 (4 performance and 4 efficiency)

  Memory: 8 GB

  System Firmware Version: 7429.61.2

  OS Loader Version: 7429.61.2

  Serial Number (system): FVFFRYP2Q6L4

  Hardware UUID: AD4F20D8-BA35-50CF-B00C-16FD4BE2DF7C

  Provisioning UDID: 00008103-000A54E802F1001E

  Activation Lock Status: Enabled

dh1tw, 2021-12-25:
There is a discussion about Github adding eventually M1 Macs to the Github Action farm - although they are still vague by when this will happen. I just downloaded the provided `sdrpp_macos_m1.zip` and it works like a charm. Thanks for this awesome piece of software and your effort of porting it into the Mac world!

AlexandreRouma, 2021-12-25:
Would be great if they did!
I think the main issue right now is the github action runners are written in C# (dotnet core) and that's not supported on M1 yet.

KZeronimo, 2021-12-25:
@AlexandreRouma let me know if you want help with the github actions .NET (aka dotnet core) will run fine on M1 (either native or thru Rosetta 2)

AlexandreRouma, 2021-12-25:
If you can get the github runner to work on a M1 machine and compile native M1 code, let me know
So far I haven't even tried since microsoft says it doesn't work

ra1nb0w, 2021-12-27:
You can easily install SDRPlusPlus with [macports](https://ports.macports.org/port/SDRPlusPlus/details/)

AlexandreRouma, 2021-12-27:
I can only garantee builds from the releases & nightly builds.
The builds I've seen in package managers tend to be built and configured completely wrong resulting in configuration issues and/or missing modules.

ra1nb0w, 2021-12-27:
You can see the configuration at https://github.com/macports/macports-ports/blob/master/science/SDRPlusPlus/Portfile
Feel free to say what is wrong.

Gymnae, 2022-01-28:
> Here is a native M1 build I just made. (ignore the previous version I posted which was broken) [sdrpp_macos_m1.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7776043/sdrpp_macos_m1.zip)

Could you create another nightly with your server support? The macports version is old now and the intel nighlty won't start on apple silicon

ra1nb0w, 2022-01-29:
FYI on macports:
- SDR++ is at the latest stable. We generally track only stable point. You can build master easily when you have installed the stable.
- sdrplay driver support for M1: I am still waiting sdrplay dev that had promise to unify the installer and release the new  stable version.

Gymnae, 2022-01-29:
> * You can build master easily when you have installed the stable.

Cool, how could I do that? I'm only interested for now in being able to connect to the beta server

ra1nb0w, 2022-01-29:
```
sudo port install sdrplusplus
# somewhere
git clone https://github.com/AlexandreRouma/SDRPlusPlus.git
cd SDRPlusPlus
mkdir build && cd build
cmake -DOPT_OVERRIDE_STD_FILESYSTEM=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON ..
```

Gymnae, 2022-01-29:
Tried that, fails with missing "volk", the one available in macports doesn't do anything when installed and brew requires building from source, which apparently also fails

ra1nb0w, 2022-01-29:
Maybe, you have conflicts between macports and homebrew. I build SDR++ without any trouble with macports.
Can you check if you have the following file `/opt/local/share/cmake/volk/VolkConfig.cmake`?
Can you post the error?

AlexandreRouma, 2022-01-29:
I'd suggest removing everything (besides the modified libvolk), starting clean and doing exactly how the CI does it:
https://github.com/AlexandreRouma/SDRPlusPlus/blob/260651fb5c328346044ec6562f7d590c220dc203/.github/workflows/build_all.yml#L77
Of course you can skip the steps about hardware you don't need.

Gymnae, 2022-01-29:
Yeah volk won't build for arm for me: 
```
Undefined symbols for architecture arm64:
  "_GetAarch64FeaturesEnumName", referenced from:
      _main in list_cpu_features.c.o
  "_GetAarch64FeaturesEnumValue", referenced from:
      _main in list_cpu_features.c.o
  "_GetAarch64Info", referenced from:
      _main in list_cpu_features.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cpu_features/list_cpu_features] Error 1
make[1]: *** [cpu_features/CMakeFiles/list_cpu_features.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 89%] Built target volk
```

AlexandreRouma, 2022-01-29:
that's normal, cpu_features, used by volk doesn't support M1 yet so you have to replace the version in the volk codebase with a fork that supports M1.

Gymnae, 2022-01-29:
Hmk, so I will hunt for one in its network

AlexandreRouma, 2022-01-29:
this one iirc
https://github.com/google/cpu_features/pull/204

NH7L, 2022-01-29:
Progress report, clueless user division. Keep in mind that I am a software ignoramus, knowing absolutely nothing about coding, compiling, dependencies or any of that stuff: 

A month later, I am very happily running the version of SDR++ Alexandre gave me (see Dec. 25, above) on two Apple-silicon machines, both bottom-of-the-line  2020 MacBook Air M1 8GB/256GB. I use SDR++ and an Airspy YouLoop to run an RTL-SDR.com v.3 and an Airspy HF+ Discovery. Though it is not listed in the SDR++ source list, I also tried my 10-year-old Funcube Dongle Pro+ just for kicks. Unsurprisingly, the app would not recognize it. How much SDR++ use? Daily, sometimes for hours. Any crashes? Yes. One, for unknown reasons. 

SDR++ is excellent. There is no other word for it. Sincere gratitude to Alexandre for his expertise and all the work he has put into (and continues to put into) this app. This is what Mac-using radio enthusiasts have been waiting for ... for years. 

Ernie NH7L

Gymnae, 2022-01-29:
> this one iirc [google/cpu_features#204](https://github.com/google/cpu_features/pull/204)

Ok, that is a bit over my head :)
I don't really know how to add this cpu_features to the cloned volk repo and compile

ra1nb0w, 2022-01-29:
Volk builds fine on macports from July 2021 (https://github.com/macports/macports-ports/commit/e83a55ef196d4283be438c052295b2fc44f3df5b and https://ports.macports.org/port/volk/details/)

ra1nb0w, 2022-01-29:
removing/moving homebrew and using only macports must build SDR++.
Otherwise, you can edit the portfile
```
sudo port edit sdrplusplus
```
changing the version with the commit and update `checksums` with
```
sudo port -d checksum sdrplusplus
```
then you can install as always

Gymnae, 2022-01-29:
The weird thing is that installing volk appears to do nothing: 
```
sudo port install cmake
--->  Computing dependencies for cmake
--->  Cleaning cmake
--->  Scanning binaries for linking errors
--->  No broken files found.
--->  No broken ports found.
```
I removed all build related thing from brew and installed cmake via ports. Still, volk is not being found by cmake

ra1nb0w, 2022-01-29:
```
port contents volk
```
what returns? You should have `.cmake` files. otherwise
```
sudo port install volk
```

Gymnae, 2022-01-29:
```
$ port contents volk
Port volk contains:
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/__init__.py
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/__init__.pyc
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/__init__.pyo
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/cfg.py
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/cfg.pyc
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/cfg.pyo
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/volk_modtool_generate.py
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/volk_modtool_generate.pyc
  /opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/volk_modtool/volk_modtool_generate.pyo
  /opt/local/bin/list_cpu_features
  /opt/local/bin/volk-config-info
  /opt/local/bin/volk_modtool
  /opt/local/bin/volk_profile
  /opt/local/include/cpu_features/cpu_features_cache_info.h
  /opt/local/include/cpu_features/cpu_features_macros.h
  /opt/local/include/cpu_features/cpuinfo_aarch64.h
  /opt/local/include/volk/constants.h
  /opt/local/include/volk/saturation_arithmetic.h
  /opt/local/include/volk/volk.h
  /opt/local/include/volk/volk_16i_32fc_dot_prod_32fc.h
  /opt/local/include/volk/volk_16i_branch_4_state_8.h
  /opt/local/include/volk/volk_16i_convert_8i.h
  /opt/local/include/volk/volk_16i_max_star_16i.h
  /opt/local/include/volk/volk_16i_max_star_horizontal_16i.h
  /opt/local/include/volk/volk_16i_permute_and_scalar_add.h
  /opt/local/include/volk/volk_16i_s32f_convert_32f.h
  /opt/local/include/volk/volk_16i_x4_quad_max_star_16i.h
  /opt/local/include/volk/volk_16i_x5_add_quad_16i_x4.h
  /opt/local/include/volk/volk_16ic_convert_32fc.h
  /opt/local/include/volk/volk_16ic_deinterleave_16i_x2.h
  /opt/local/include/volk/volk_16ic_deinterleave_real_16i.h
  /opt/local/include/volk/volk_16ic_deinterleave_real_8i.h
  /opt/local/include/volk/volk_16ic_magnitude_16i.h
  /opt/local/include/volk/volk_16ic_s32f_deinterleave_32f_x2.h
  /opt/local/include/volk/volk_16ic_s32f_deinterleave_real_32f.h
  /opt/local/include/volk/volk_16ic_s32f_magnitude_32f.h
  /opt/local/include/volk/volk_16ic_x2_dot_prod_16ic.h
  /opt/local/include/volk/volk_16ic_x2_multiply_16ic.h
  /opt/local/include/volk/volk_16u_byteswap.h
  /opt/local/include/volk/volk_16u_byteswappuppet_16u.h
  /opt/local/include/volk/volk_32f_64f_add_64f.h
  /opt/local/include/volk/volk_32f_64f_multiply_64f.h
  /opt/local/include/volk/volk_32f_8u_polarbutterfly_32f.h
  /opt/local/include/volk/volk_32f_8u_polarbutterflypuppet_32f.h
  /opt/local/include/volk/volk_32f_accumulator_s32f.h
  /opt/local/include/volk/volk_32f_acos_32f.h
  /opt/local/include/volk/volk_32f_asin_32f.h
  /opt/local/include/volk/volk_32f_atan_32f.h
  /opt/local/include/volk/volk_32f_binary_slicer_32i.h
  /opt/local/include/volk/volk_32f_binary_slicer_8i.h
  /opt/local/include/volk/volk_32f_convert_64f.h
  /opt/local/include/volk/volk_32f_cos_32f.h
  /opt/local/include/volk/volk_32f_exp_32f.h
  /opt/local/include/volk/volk_32f_expfast_32f.h
  /opt/local/include/volk/volk_32f_index_max_16u.h
  /opt/local/include/volk/volk_32f_index_max_32u.h
  /opt/local/include/volk/volk_32f_invsqrt_32f.h
  /opt/local/include/volk/volk_32f_log2_32f.h
  /opt/local/include/volk/volk_32f_null_32f.h
  /opt/local/include/volk/volk_32f_s32f_32f_fm_detect_32f.h
  /opt/local/include/volk/volk_32f_s32f_add_32f.h
  /opt/local/include/volk/volk_32f_s32f_calc_spectral_noise_floor_32f.h
  /opt/local/include/volk/volk_32f_s32f_convert_16i.h
  /opt/local/include/volk/volk_32f_s32f_convert_32i.h
  /opt/local/include/volk/volk_32f_s32f_convert_8i.h
  /opt/local/include/volk/volk_32f_s32f_mod_rangepuppet_32f.h
  /opt/local/include/volk/volk_32f_s32f_multiply_32f.h
[... 102 lines cut ...]
  /opt/local/lib/cmake/CpuFeatures/CpuFeaturesConfigVersion.cmake
  /opt/local/lib/cmake/CpuFeatures/CpuFeaturesTargets-release.cmake
  /opt/local/lib/cmake/CpuFeatures/CpuFeaturesTargets.cmake
  /opt/local/lib/libcpu_features.a
  /opt/local/lib/libvolk.2.5.0.dylib
  /opt/local/lib/libvolk.2.5.dylib
  /opt/local/lib/libvolk.dylib
  /opt/local/lib/pkgconfig/volk.pc
  /opt/local/share/cmake/volk/VolkConfig.cmake
  /opt/local/share/cmake/volk/VolkConfigVersion.cmake
  /opt/local/share/cmake/volk/VolkTargets-release.cmake
  /opt/local/share/cmake/volk/VolkTargets.cmake
```

But when I run 
`cmake -DOPT_OVERRIDE_STD_FILESYSTEM=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON .. `(from a previous comment) I get:
```
-- Checking for module 'volk'
--   No package 'volk' found
CMake Error at /opt/local/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  core/CMakeLists.txt:83 (pkg_check_modules)
```

ra1nb0w, 2022-01-29:
you have something strange. I just tested sdrplusplus trunk on M1 mini and macOS 11 and it works fine without error.
maybe some CXX/CC flags in the environment?
Ps. the volk installation is fine and all files are in the right place.

Gymnae, 2022-01-29:
Maybe :) I don't even know that these CXX/CC flags are and where they would have been set

ra1nb0w, 2022-01-29:
maybe homebrew. how know.
You should use the `port edit` instruction

Gymnae, 2022-01-30:
Ah, I think I will just wait for stable and continue with openwebrx until then :)

ra1nb0w, 2022-01-30:
ok

Gymnae, 2022-01-30:
I do really appreciate the attempts to help me out here, maybe there's something odd with my system. I am looking forward to ditching openwebrx for local listening once I can connect to the sdrplusplus server

madjuju, 2022-12-01:
Hello 
I'm willing to spend some time compiling the release on my Mac Studio 
I have tried several times on my side but I never succeeded.

I use xcode every day.
I refuse macport because it compiles randomly and generates all the dependencies whereas hombrew is the best solution.

If you know how to give me a guide to make nice compilations I can do it regularly and push it @AlexandreRouma

kevinkchoi, 2023-01-28:
Hi Everyone.
Happy new year to you all, specially to Alexandre, making SDR++.
I'm new here and just bought an RTL-SDR dongle... and as you guess, I'm stuck with SDR++ installation on my macbook m1.
I tried to follow your instruction above, but completely lost.
Homebrew installed but somehow got error on "-march=native" error. 
And 3rd attempt gave me another error, that I don't understand.
==================
    cmake .. -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=ON -DOPT_BUILD_AUDIO_SINK=OFF  -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=ON -DUSE_BUNDLE_DEFAULTS=ON -DCMAKE_BUILD_TYPE=Release
  CMake Deprecation Warning at core/libcorrect/CMakeLists.txt:1 (cmake_minimum_required):
     Compatibility with CMake < 2.8.12 will be removed from a future version of
     CMake.
  
    Update the VERSION argument <min> value or use a ...<max> suffix to tell
    CMake that the project does not need compatibility with older versions.

  -- Checking for module 'glfw3'
  --   No package 'glfw3' found
  CMake Error at /opt/local/share/cmake-3.24/Modules/FindPkgConfig.cmake:607 (message):
    A required package was not found
  Call Stack (most recent call first):
    /opt/local/share/cmake-3.24/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
    core/CMakeLists.txt:59 (pkg_check_modules)

   -- Configuring incomplete, errors occurred!
   See also "/Users/kevinchoi/SDRPlusPlus/build/CMakeFiles/CMakeOutput.log".
   See also "/Users/kevinchoi/SDRPlusPlus/build/CMakeFiles/CMakeError.log".
=========================

Macports, well, stuck somehow.

Could someone help me to have a clean step-by-step installation procedure please?

kevinkchoi, 2023-01-28:
Alexandre,
In addition to my comment here above, I also tried to run your build version of sdrpp_macos_m1.zip.
It generates error of execution saying;

"SDR++" can't be opened because Apple cannot check it for malicious software.
This software needs to be updated. Contact the developer for more information..

Please help....

michaelgeorge001, 2023-01-28:
Hi,

These Instructions are from Apple -

1. In the Finder  on your Mac, locate the app you want to open. [Do not use Launchpad to locate the app.]

2. Control-click the app icon, then choose Open from the shortcut menu.
 
3. Click Open. [The app is saved as an exception to your security settings, and you can open it at any time by double-clicking it, just as you do with any registered app.]

AlexandreRouma, 2024-02-11:
An apple silicon build is now available: https://github.com/AlexandreRouma/SDRPlusPlus/releases/tag/nightly
The file is `sdrpp_macos_arm.zip`

closed 2024-02-11 by AlexandreRouma