AlexandreRouma/SDRPlusPlus #565: Request: AFEDRI AFE822x SDR-Net

afesan, 2021-12-29:
Could be added to work with SDR++? 
Would be great!
Thanks in advance.
Tony

AlexandreRouma, 2021-12-29:
I don't own any afedri SDR so I can'tcurrently  implement support.
If the company is willing to send a development unit then I would be able to next febuary.

gvanem, 2022-01-04:
I build the yesterday's master with the updated `rfspace_source.dll`. Tried it with my Afedri-LAN receiver.
But SDRpp crashed on connect inside some Json code I fail to understand:
```
rfspace_source!RFSpaceSource::deviceInit+0x585:
616af2c5 8b38            mov     edi,dword ptr [eax]  ds:002b:00000000=?????

        bool changed = false;
        config.acquire();
        if (!config.conf["devices"].contains(devConfName)) {
            config.conf["devices"][devConfName]["sampleRate"] = sampleRates.key(0); << ! here
            config.conf["devices"][devConfName]["gain"] = 0;
```
The generated `rfspace_config.json` only has this:
```
{
    "devices": {},
    "hostname": "10.0.0.50",
    "port": 50000
}
```
No `"samplerate"` !?

klikis, 2022-01-04:
idem

AlexandreRouma, 2022-01-04:
why would you expect a Afedri-LAN SDR to run with a beta source designed for RFspace devices (CloudIQ, CloudSDR, NetSDR( (not yet supported), SDR-IP (not yet supported))?

AlexandreRouma, 2022-09-13:
Looks like support for it will likely not be added

closed 2022-09-13 by AlexandreRouma