AlexandreRouma/SDRPlusPlus #566: Memory Naming of Issues

davemc1978, 2021-12-29:
Hello,

Would it be possible to change how the memory naming function works when trying to save different frequencies with the same name?

For example, I would like to save MW stations, some of which have the same name but broadcast on different frequencies, at the moment only the first entry of the same name is saved, if I try to save a different frequency, but name it the same it won't work.

A work around I use is to increment the original name by one each time, for example TalkSport, TalkSport1, TalkSport2 etc which gets the job done but is a bit clunky.

It would be great if it would be possible to use the same name multiple times!

Many thanks for the great application, I enjoy using it a lot!

Many thanks
Dave

AlexandreRouma, 2021-12-29:
Hello,
Unfortunately this is not possible without a complete rewrite of the frequency manager.
The name serves as the ID of the bookmark and having two with the same name would completely break the way they're saved and access. (Just like having two files with the same name in a directory doesn't make much sense)

davemc1978, 2021-12-31:
Hello,

Many thanks for taking the time to get back to me, and for the information regarding the memory manager.  This is a shame as this will effect memory saving for stations in the MW, SW and FM broadcast bands which I listen too a lot.  However, I have a work around which still achieves what I'm trying to do, so it's not a major problem.

Thanks for all your time and efforts in developing the app, it's really appreciated.

Kind regards
Dave

tomswift837, 2022-01-03:
Dave,
I faced the same situation.  I worked around it by putting a "space" at the end of the 2nd entry.  If I had more "same named" entries, I would just add more blank spaces.  The program will not let you over write entries, so you are safe there.

starvald, 2022-01-03:
you can have the same names as long as they are in separate lists

closed 2022-09-29 by AlexandreRouma