AlexandreRouma/SDRPlusPlus #572: CW mode very choppy

clytle374, 2022-01-01:
Running the latest git source, be this way for a week or more.  When selecting CW mode the display and audio drop to about one frame per second.  No issues on other modes.  Have to work the next few day, might be a while before I can respond to questions.  

Also I have channels ghosting in sdr++ that I don't have in gqrx.  Any suggestions?
Thanks, Cory

AlexandreRouma, 2022-01-01:
your computer can't handle the bandwidth you've selected for the CW demodulatot

clytle374, 2022-01-01:
Thanks, there are 3 bandwidths.  Sadly I don't understand what is what there. 
The first is directly under the limesdr device.  Lowering way down helps it.
Another bandwidth a few more lines down.  If I don't keep this way up, I lose most of my signals. This doesn't affect hangs.
The one below CW selection maxes out at 500, I have to keep it up around 500 even with the top setting turned down.  This never happened until recently.
 Machine is a 6 core  Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
I've got the limesdr amplified and with a LPF now, building a new loop with more amplification.
Thanks

AlexandreRouma, 2022-01-01:
what sample rate is the LimeSDR running at?

AlexandreRouma, 2022-01-01:
Also for the "channel ghosting" thing, send a screenshot of both GQRX and SDR++
My guess is you either messed with the bandwidth setting in the source menu or have the gain too high

clytle374, 2022-01-15:
Sorry, I got sidetracked with work and other issues.  I've messed with all the settings. All 3 gains are maxed on gqrx as the limesdr receive is so weak at these frequencies.  My low pass filter has been corrected and tested, FM no longer getting into the receiver.   I don't think it's a AM broadcast station as I can't receive any currently.  

[image]
Here is WWV that I can't pick up with sdr++
[image]
Here is the screenshot with CW working, I have turned the setting down right below the limesdr device in the menu.  Before I left it a max all the time, now it works fine at max in any mode other than CW.
Building a new amplified antenna, probably going to quit fighting this until I get enough signal into the lime to work better.  
Thanks

clytle374, 2022-01-18:
Apologies for the previous post, I didn't realize until just now I didn't post the 2 pictures of the same frequencies. 

The first number is the sample rate of the Limesdr?  I brought it down to 5MS/s, at the same time I set bandwidth to auto and I got absolutely nothing. Again, I don't completely the bandwidth setting.  But setting it to low numbers renders receiver completely deaf, it's only mostly deaf in HF stock.  Samples per second from the SDR makes sense. I finally took some time to investigate more as the loop antenna amplifier project is making me crazy, lol.  

Here is what I figured out as I copied settings from gqrx and still had nothing on SDR++.  

gqrx
[image]

sdr++ on auto
[image]
 

sdr++ after manually changing bandwidth back to maximum
[image]

Advice or reading materials?  Am I doing something stupid? 
Thanks, Cory

AlexandreRouma, 2022-09-13:
this should be fixed in the nightly

closed 2022-09-13 by AlexandreRouma