AlexandreRouma/SDRPlusPlus #573: SDR++ version 1.04 server issue connecting to WSJT-X version 2.5.3.

W5QJ, 2022-01-01:
Have not been able to successfully connect to rig control using a RTL-SDR v3.  Hamlib library installed is w64 version 4-4.  Local host and port is 127.0.0.1:4532.  SDR++ Rigctl Server is running and initially shows "Connected" to WSJT-X.  WSJT-X reports connection error and after re-entering radio control parameters, reports hamlib error.  SDR++ Rigclt Server then reverts to "listening".
[image]
[image]
Same error occurs with or without firewall being active.

Thanks

AlexandreRouma, 2022-01-01:
Use an older wsjt-x version, looks like they broke their Net RigCTL client in the latest version or the rigctl protocol got breaking changes I don't know about.
I don't have the time to fix this stuff at the moment.

ja-Bertolin, 2023-03-13:
At least with GQRX (version April 2022) works fine, so I guess that they did not break their Net RigCTL client. Should be another reason related to SDR++

KAHHN, 2023-03-17:
Good day, great work on all your work thank you, so I am running sdr++ on my laptop and I want to start using my tab A ,android and cant find the the sdr++ .exe for  my android.
 I am sure you are busy, however if you could poi t me in the correct direction I would appreciate that a lot ....

x31415926, 2023-04-08:
> At least with GQRX (version April 2022) works fine, so I guess that they did not break their Net RigCTL client. Should be another reason related to SDR++

I run into this as well and did a bit of spelunking and while I don't quite understand the correct/expected behavior, I could at least explain what is happening.

* Hamlib offers a vfo mode which, when enabled, will result in sending commands that include a vfo string.
* Hamlib netrigctl (and as far as I see netrigctl only), which is what is used in the wsjt-x/sdrpp setup, has an additional vfo option that signals whether rigtctl was started with the vfo option.
* The code in netrigctl however checks for either of them when deciding to include this string
* This specific behavior was introduced with the fix for https://github.com/Hamlib/Hamlib/issues/628. However, before that the code checked for vfo option, which interestingly, is always set when the server responds to the chk_vfo command (and is the vfo mode that is actually read from the 0|1 response)
* Net-net, the result of sdrpp responding with "CHK_VFO 0" is the introduction of the vfo string in the set_freq command, which, in turn, breaks sdrpp parsing.

Me thinks this is a hamlib bug (namely, the library should check for the vfo mode only, not for the OR between vfo option/vfo mode). But I am just someone reading code and the man page for rigctl, without a clear understanding of rigctl features... Alternatively (and this is actually what GQRX code does) this could be worked around in sdrpp with this patch:

```
~/git/SDRPlusPlus$ git diff
diff --git a/misc_modules/rigctl_server/src/main.cpp b/misc_modules/rigctl_server/src/main.cpp
index 801bec3..21bcdb0 100644
--- a/misc_modules/rigctl_server/src/main.cpp
+++ b/misc_modules/rigctl_server/src/main.cpp
@@ -552,7 +552,7 @@ private:
         }
         else if (parts[0] == "\\chk_vfo") {
             std::lock_guard lck(vfoMtx);
-            resp = "CHKVFO 0\n";
+            resp = "RPRT 1\n";
             client->write(resp.size(), (uint8_t*)resp.c_str());
         }
         else if (parts[0] == "s") {
```

Which basically tells rigctl the vfo option is not supported, resulting in the client not sending any VFO as a result. I tested this and it seems to work with the latest sdrpp and wesjt-x 2.6.1.

Anyway, I will let the sdrpp owner decide if he wants to workaround it or not. If I find the time to dig more I might open an issue in hamlib...

AlexandreRouma, 2023-04-08:
To me, responding with `CHKVFO 0` should tell hamlib that the mode isn't supported at all instead of just responding with an error report.
So yes imo it's an hamlib bug which should be reported.
As for the work around I guess I can implement it but I don't want this deterring from hamlib fixing the bug.

hellmind, 2023-05-18:
I fix my rigctl  upgrading hamlib from 
rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
to
rigctl Hamlib 4.6~git Apr 29 17:56:00Z 2023 SHA=9318df 64-bit

gprime31, 2023-07-21:
Has this been fixed? on linux and windows.

x31415926, 2023-07-22:
The issue I opened on hamlib was marked fixed. I installed WSJT-X rc2 and connecting to SDR++ seems to function as expected.

closed 2024-01-26 by AlexandreRouma