AlexandreRouma/SDRPlusPlus #574: How do i explain this...

palWorx, 2022-01-02:
**Standard Intel Hardware x86_64 & SDRplay RSP1a
Debian 11 stable (up to date || 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux)
SDR++ 1.05 (Build 04:55 Jan 01 2022)
IF AGC (Parameters untouched) and IQ Correction is on
Bandwith 2-6 MHz (makes no difference)
ZeroIF (makes no difference)

Problem exists since 1.0.1**

If I change the frequency to 234 khz (RTL Luxembourg) I have to set the LNA Gain of the RSP1a to maximum and raise the bottom of the FFT (via min). So far normal behavior.

Almost always (sometimes it works) the FFT is dead, even the huge spike at 0 khz is missing. Then I have to shift everything to the left until 0 khz has moved out of the field of view. Only then I receive something, the FFT comes to life and I also get an audio signal.

It also does not work if I (before I start the RSP1a via the play button) select the frequency 234 khz and then start the receiver. Tried it with and without zoom.

AlexandreRouma, 2022-01-02:
I don't understand your issue. Can you screen record it happening and explain the expected behavior?

palWorx, 2022-01-02:
I try to explain the expexted behavior first:
When i change from another frequency to 234khz or lets say 77khz i expect to hear the transmitter like DCF77 ort RTL. Instead there is no reception, no signal on the FFT and no audio beside static. I tried to do some screenshots but now i am not even able to receive 234khz anymore (tried the last 10 Minutes, with restart of SDR++ and unplugging RSP1a,...).With other software GQRX for example i hav no problem to receive mentioned frequencies.
You made it possible to grab the part of the FFT (middle of the screen) where the frequencies are shown with the mouse (holding the left mouse trigger) and drag the screen in a given direction. When i did this in the past an moved the 0khz and the shown "negative" frequency part to the left an out of the view so that lets say 100khz an up are shown all came to life before.

I am out of words. If you still don't understand me, i'm sorry for all the fuss. In this case close the Ticket.

AlexandreRouma, 2022-01-02:
By disappearing I'm gonna guess the signal level is going too low and you have to change the "min" parameter of the waterfall to see it again.

But again, I'd like a screenshot to see exactly what you see, it doesn't matter if the signal doesn't show, I just want to see exactly what's going on.

palWorx, 2022-01-02:
Don't know why, but 234khz is back again in SDR++. Change min and max out the LNA Gain is normal procedere below 30MHz.

Working (after the FFT showed the correct signals according to the expected reception i moved it, so you can see the 0KHz Spike and the "negative" frequencies i was talking about):
[image]

Not working:
[image]

AlexandreRouma, 2022-01-02:
My suspicious is that the center of the passband (when completely zoomed out) is in the negative frequencies, which the SDR cannot tune to.

Can you grab a screenshot of the debug menu when it's not working as expected?

palWorx, 2022-01-02:
Should i select any options in the debug section?

[image]

palWorx, 2022-01-02:
Another non-working:
[image]

AlexandreRouma, 2022-01-02:
The first one as expected has the center of the passband in the negative frequencies.
And the second one has the frequency your want to receive on the far edge of the passband. You receive signals that close to the edge of the passband with SDRplay & other common SDRs.

palWorx, 2022-01-02:
Sorry i dont get it. The first one in tthe negative freqeuncies i understand. With the rest i dont understand what you are trying to tell me.
I no ham radio pro, i just like to listen to some frequencies.

palWorx, 2022-01-02:
A little Google brought this about Passband/Bandpass-Filter: https://en.wikipedia.org/wiki/Passband
Far edge off 0-2MHz Bandpass-Filter according to these Pictures but i still dont understand what this means. Why is it working in GQRX?

[image]
[image]

AlexandreRouma, 2022-01-02:
Either GQRX hides the sides (which I don't think it does) or you're not putting the VFO on the far edge of the passband.

And by passband I don't mean input filter, I mean IQ passband, aka what you can see on the waterfall. Look at the sides of the waterfall when zoomed out. You'll notice the level of the signal drops off significantly. This is normal behavior for most SDRs. It doesn't matter at which frequency you're tuned to, the signal level on the sides will be attenuated.

palWorx, 2022-01-02:
Now i understand what you mean and no, thats not the case. What you see in the non working picture is more of cosmetic nature, because i overcomplicated my explanation regarding the 0KHz and below - so i shifted it for the screenshot. I had it before moved to different sides with no luck. Yeah, i know i am no easy one.

At first i am within the range with no reception and after moving all to the left and zooming it i receive RTL and DCF77.

Listen, obviously this is a wontfix or from your perspective expected behavior and i am gratefull for the time you invested. I can live with it and will enjoy SDR++ non the less, i can only repeat from another bugreport: SDR++ is the best SDR-Software under linux.

Thanks and keep going!

closed 2022-01-02 by palWorx