AlexandreRouma/SDRPlusPlus #575: compilation error

gariac, 2022-01-03:
I'm running opensuse 15.2 so I am building from source. I did the usual git clone then in that directory made a build director then cmake ..

First attempt at compilation:
-----------------

[ 63%] Building CXX object source_modules/airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o
/usr/local/src/SDRPlusPlus/source_modules/airspy_source/src/main.cpp:10:10: fatal error: airspy.h: No such file or directory
   10 | #include <airspy.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [source_modules/airspy_source/CMakeFiles/airspy_source.dir/build.make:80: source_modules/airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1526: source_modules/airspy_source/CMakeFiles/airspy_source.dir/all] Error 2
make: *** [Makefile:147: all] Error 2
linux-yxjo:/usr/local/src/SDRPlusPlus/build #
-----------------------------------------------------
I have airspy.h in /usr/include/libairspy. I'm not sure why it isn't found. 

I made the following edit to CMakeCache.txt:
LIBAIRSPY_INCLUDE_DIRS:INTERNAL=/usr/include/;/usr/include/libusb-1.0/;/usr/include/libairspy
LIBAIRSPY_CFLAGS:INTERNAL=-I/usr/include/;-I/usr/include/libusb-1.0;-I/usr/include/libairspy

You can see that error is gone but a new one occurs:
---------------------------------------------------
[ 63%] Building CXX object source_modules/airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o
/usr/local/src/SDRPlusPlus/source_modules/airspy_source/src/main.cpp: In member function ‘void AirspySourceModule::refresh()’:
/usr/local/src/SDRPlusPlus/source_modules/airspy_source/src/main.cpp:82:17: error: ‘airspy_list_devices’ was not declared in this scope; did you mean ‘airspy_device’?
   82 |         int n = airspy_list_devices(serials, 256);
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 airspy_device
make[2]: *** [source_modules/airspy_source/CMakeFiles/airspy_source.dir/build.make:80: source_modules/airspy_source/CMakeFiles/airspy_source.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1526: source_modules/airspy_source/CMakeFiles/airspy_source.dir/all] Error 2
make: *** [Makefile:147: all] Error 2

gvanem, 2022-01-12:
> error: ‘airspy_list_devices’ was not declared

It's [here](https://github.com/airspy/airspyone_host/blob/master/libairspy/src/airspy.h#L126). Why don't you update it? And fix the awful markdown formatting in your issue.

AlexandreRouma, 2022-01-12:
this issue happens when you have an outdated version of libairspy.
Update to the latest libairspy version and try again

aosync, 2022-08-17:
I'll close this as this entry has staled after a suggestion has been given, feel free to reopen.

closed 2022-08-17 by aosync