AlexandreRouma/SDRPlusPlus #578: SDRpp is trying to find things I don't want

loewal, 2022-01-05:
Hello Alexandre,
First of all: a happy new year!

I compiled from source and set the necessary modules to OFF.
All went fine, but when I start sdrpp from the command promt I get this:

[2022-01-05 17:42:35.946] [info] Initializing Airspy Source (airspy_source)
[2022-01-05 17:42:35.956] [error] Module 'airspy_source' doesn't exist
[2022-01-05 17:42:35.956] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-05 17:42:35.971] [error] Module 'airspyhf_source' doesn't exist
[2022-01-05 17:42:35.971] [info] Initializing Audio Sink (audio_sink)
[2022-01-05 17:42:35.986] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-05 17:42:36.001] [error] Module 'bladerf_source' doesn't exist
[2022-01-05 17:42:36.001] [info] Initializing File Source (file_source)
[2022-01-05 17:42:36.016] [error] Module 'file_source' doesn't exist
[2022-01-05 17:42:36.016] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-05 17:42:36.031] [info] Initializing HackRF Source (hackrf_source)
[2022-01-05 17:42:36.056] [info] Initializing LimeSDR Source (limesdr_source)
[2022-01-05 17:42:36.060] [error] Module 'limesdr_source' doesn't exist
[2022-01-05 17:42:36.060] [info] Initializing Network Sink (network_sink)
[2022-01-05 17:42:36.076] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-05 17:42:36.091] [error] Module 'plutosdr_source' doesn't exist
[2022-01-05 17:42:36.091] [info] Initializing RFspace Source (rfspace_source)
[2022-01-05 17:42:36.105] [error] Module 'rfspace_source' doesn't exist
[2022-01-05 17:42:36.105] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-01-05 17:42:36.129] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-05 17:42:36.135] [error] Module 'rtl_tcp_source' doesn't exist
[2022-01-05 17:42:36.135] [info] Initializing Radio (radio)
[2022-01-05 17:42:36.194] [info] FM IF Noise reduction set to 32 taps
[2022-01-05 17:42:36.241] [info] RtAudio stream open
[2022-01-05 17:42:36.241] [info] Initializing Recorder (recorder)
[2022-01-05 17:42:36.242] [error] Module 'recorder' doesn't exist
[2022-01-05 17:42:36.242] [info] Initializing Rigctl Server (rigctl_server)
[2022-01-05 17:42:36.243] [error] Module 'rigctl_server' doesn't exist
[2022-01-05 17:42:36.243] [info] Initializing SDRplay Source (sdrplay_source)
[2022-01-05 17:42:38.814] [error] Could not init RSP device: sdrplay_api_Fail
[2022-01-05 17:42:38.814] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Using device #0: Mirics MSi2500 default (e.g. VTX3D card)
usb_claim_interface error -6
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to open mirisdr device.
Aborted (core dumped)

**Can you give me a clue? The errors are about modules I set to OFF in cmakelist**

SoapySDRUtil shows this:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g6f97389b
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so  (0.3.3-8d2e7be)
Module found: /usr/local/lib/SoapySDR/modules0.8/libLMS7Support.so    (20.07.2-28031bfc)
Module found: /usr/local/lib/SoapySDR/modules0.8/libaudioSupport.so   (0.1.2-79129c9)
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so  (0.3.2-53ee8f4)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-b2d2a17)
Module found: /usr/local/lib/SoapySDR/modules0.8/libuhdSupport.so     (0.4.1-1f7b6fa)
Available factories... audio, hackrf, lime, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
I am lost....

Can you help?

AlexandreRouma, 2022-01-05:
Hello,
This is normal and can be ignored. SDR++ doesn't care which modules were enabled when compiling when creating it's default config. This means it'll try to load modules that might not be present. This is done on purpose so that if a user later decides to compile the missing modules, they won't have to mess with the module manager to enable the modules, or even worse, have to recompile the entire software because when resetting the config, it doesn't try to load the modules.

AlexandreRouma, 2022-01-05:
oops clicked the wrong button.
The other issue:
The issue you're getting with soapy is a soapy issue unrelated to SDR++. Try `SoapySDRUtil --find` and you'll likely get the same error

loewal, 2022-01-05:
Helas,
SoapySDRUtil —find or probe=“driver=sdrplay” is okay. Under SDRangel the rsp1 runs fine.
Why mirisdr?

AlexandreRouma, 2022-01-05:
I don't know if it uses the same installed instance. 
mirisdr is the old name of the sdrplay devices, named after the Mirics chipset they use.
I don't know why soapy crashes when loading (especially since you haven't given your which OS you're under and what version of SDR++ you're using, along with where you got it from, aka, from a deb I published, built from source or package manager) but I'd suggest disabling soapysdr and using your RSP1 with the dedicated SDRplay source.

AlexandreRouma, 2022-01-05:
From the USB error btw, it just seems like something else is already using a device that soapy is trying to access. Aka probably another SDR++ source module.

AlexandreRouma, 2022-09-13:
closing as idle

closed 2022-09-13 by AlexandreRouma