AlexandreRouma/SDRPlusPlus #579: segment violation on Ubuntu 20.04

Edwrom, 2022-01-05:
Version :  SDR++ v1.0.3
Elementary OS, (ubuntu 20.04)
amdgpu drivers., all updates to date. 
SDR++ install from .deb release
RTL-SDR connected and worked on CubicSDR, Gqrx and Quisc.

`sdrpp
[2022-01-05 16:59:35.626] [info] SDR++ v1.0.4
[2022-01-05 16:59:35.650] [info] Loading config
[2022-01-05 16:59:35.947] [info] Using OpenGL 3.0
Violación de segmento / segment violation
`

AlexandreRouma, 2022-01-05:
First of all, your version is out of date, get the latest nightly build and check if the issue is still present: https://www.sdrpp.org/nightly

Secondly, how much memory does your computer have? if sdr++ runs out of ram it'll segfault like that

Edwrom, 2022-01-06:
Hi, I download and install and this appear at execute sdrpp: `sdrpp 
[2022-01-05 20:40:00.547] [info] SDR++ v1.0.5
[2022-01-05 20:40:00.547] [info] Loading config
[2022-01-05 20:40:00.547] [info] Missing key in config fullscreen, repairing
[2022-01-05 20:40:00.547] [warning] ConfigManager locked, waiting...
[2022-01-05 20:40:00.758] [info] Using OpenGL 3.0
Violación de segmento/  segment violation `

Edwrom, 2022-01-06:
This: [https://www.asus.com/co/Laptops/For-Home/All-series/ASUS-M415/techspec/](url)
`OS: elementary OS 6.1 Jólnir x86_64 
ASUSLaptop X415
 Kernel: 5.11.0-43-generic 
  Uptime: 10 hours, 29 mins 
Packages: 4028 (dpkg), 34 (flatpak), 11 (snap) 
Shell: bash 5.0.17 
Resolution: 1920x1080 
DE: Pantheon 
 WM: Mutter(Gala) 
Theme: io.elementary.stylesheet.blueberry [GTK3] 
Icons: elementary [GTK3] 
Terminal: io.elementary.t 
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz 
GPU: AMD ATI 02:00.0 Picasso 
Memory: 4233MiB / 15903MiB `

AlexandreRouma, 2022-01-06:
The only thing that could be causing this is the graphics driver imo. The GPU itself should be fine.
Try updating the graphics drivers

Edwrom, 2022-01-06:
**Test in ubuntu 20.04 live usb same computer **
```
glxinfo | grep 'version'
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.60
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL shading language version string: 4.60
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions,
```
At open Sdrpp:
`sdrpp
[2022-01-06 18:06:39.505] [info] SDR++ v1.0.4
[2022-01-06 18:06:39.505] [info] Loading config
[2022-01-06 18:06:39.645] [info] Using OpenGL 3.0
[2022-01-06 18:06:39.763] [info] Loading icons
[2022-01-06 18:06:39.772] [info] Loading band plans
[2022-01-06 18:06:39.774] [info] Loading band plans color table
[2022-01-06 18:06:39.779] [info] Loading modules
[2022-01-06 18:06:39.779] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2022-01-06 18:06:39.781] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-01-06 18:06:39.782] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-01-06 18:06:39.796] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/soapy_source_config.json' does not exist, creating it
[2022-01-06 18:06:39.797] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-01-06 18:06:39.812] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2022-01-06 18:06:39.812] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-01-06 18:06:39.830] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-01-06 18:06:39.847] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/rtl_sdr_config.json' does not exist, creating it
[2022-01-06 18:06:39.847] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-01-06 18:06:39.863] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-01-06 18:06:39.880] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-01-06 18:06:39.898] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-01-06 18:06:39.912] [error] Couldn't load /usr/lib/sdrpp/plugins/plutosdr_source.so.
[2022-01-06 18:06:39.912] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2022-01-06 18:06:39.931] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/new_audio_sink_config.json' does not exist, creating it
[2022-01-06 18:06:39.931] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-01-06 18:06:39.946] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-01-06 18:06:39.963] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2022-01-06 18:06:39.981] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-01-06 18:06:39.996] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/limesdr_config.json' does not exist, creating it
[2022-01-06 18:06:39.996] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-01-06 18:06:40.013] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/hackrf_config.json' does not exist, creating it
[2022-01-06 18:06:40.014] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-01-06 18:06:40.029] [warning] ConfigManager locked, waiting...
[2022-01-06 18:06:40.029] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-01-06 18:06:40.047] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-01-06 18:06:40.062] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2022-01-06 18:06:40.081] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/bladerf_config.json' does not exist, creating it
[2022-01-06 18:06:40.082] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-01-06 18:06:40.100] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-01-06 18:06:40.115] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/airspyhf_config.json' does not exist, creating it
[2022-01-06 18:06:40.115] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-01-06 18:06:40.129] [warning] Config file '/home/ubuntu-mate/.config/sdrpp/airspy_config.json' does not exist, creating it
[2022-01-06 18:06:40.130] [info] Initializing Airspy Source (airspy_source)
[2022-01-06 18:06:40.153] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-06 18:06:40.169] [info] Initializing Audio Sink (audio_sink)
[2022-01-06 18:06:40.178] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-06 18:06:40.204] [error] Could not list devices -7
[2022-01-06 18:06:40.204] [info] Initializing File Source (file_source)
[2022-01-06 18:06:40.211] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-06 18:06:40.228] [info] Initializing HackRF Source (hackrf_source)
[2022-01-06 18:06:40.253] [info] Initializing LimeSDR Source (limesdr_source)
[... 95 lines cut ...]
glxinfo | grep 'version'
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.3.0-devel
OpenGL core profile shading language version string: 4.60
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.3.0-devel
OpenGL shading language version string: 4.60
[... cut ...] evel
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions, 
`
this appear with nightly build:
`sdrpp
[2022-01-06 15:27:05.343] [info] SDR++ v1.0.5
[2022-01-06 15:27:05.343] [info] Loading config
[2022-01-06 15:27:05.386] [info] Missing key in config fullscreen, repairing
[2022-01-06 15:27:05.386] [warning] ConfigManager locked, waiting...
[2022-01-06 15:27:05.729] [info] Using OpenGL 3.0
Violación de segmento
`

aeburriel, 2022-01-09:
It looks like the well known ImGui 1.83 problem with the newer OpenGL drivers. Can you show a backtrace?
See ticket #529

AlexandreRouma, 2022-09-13:
ImGUi has been updated a while ago so I assume this is fixed

closed 2022-09-13 by AlexandreRouma