AlexandreRouma/SDRPlusPlus #586: Problems building from the source on Pi3B with the latest RaspberryPi OS.

good-point-lab, 2022-01-14:
OS Kernel version: 5.10. Follow the steps in the rpi_install.sh with few changes, for example
Dependencies:
sudo apt install -y build-essential cmake git libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libairspy-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev librtlsdr-dev libbladerf-dev liblimesuite-dev p7zip-full wget

Build
cmake .. -DOPT_BUILD_LIMESDR_SOURCE=ON -DOPT_BUILD_RTL_SDR_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=OFF

After several attempts, unfortunately, can not start the application:

pi@raspberrypi:~/SDRPlusPlus/build $ sdrpp
[2022-01-13 09:31:58.194] [info] SDR++ v1.0.5
[2022-01-13 09:31:58.195] [info] Loading config
[2022-01-13 09:31:58.196] [warning] ConfigManager locked, waiting...
[2022-01-13 09:32:00.687] [info] Using OpenGL 3.0
[2022-01-13 09:32:01.980] [info] Loading icons
[2022-01-13 09:32:02.746] [info] Loading band plans
[2022-01-13 09:32:02.780] [info] Loading band plans color table
[2022-01-13 09:32:02.860] [info] Loading modules
[2022-01-13 09:32:02.860] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-01-13 09:32:02.877] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-01-13 09:32:02.888] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-01-13 09:32:02.897] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-01-13 09:32:02.941] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-01-13 09:32:02.953] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-01-13 09:32:02.964] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-01-13 09:32:02.977] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-01-13 09:32:02.989] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-01-13 09:32:02.999] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-01-13 09:32:03.026] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-01-13 09:32:03.037] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-01-13 09:32:03.048] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-01-13 09:32:03.059] [warning] ConfigManager locked, waiting...
[2022-01-13 09:32:03.059] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-01-13 09:32:03.095] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-01-13 09:32:03.108] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-01-13 09:32:03.165] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-01-13 09:32:03.177] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-01-13 09:32:03.194] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-01-13 09:32:03.206] [info] Initializing Airspy Source (airspy_source)
[2022-01-13 09:32:03.229] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-13 09:32:03.249] [info] Initializing Audio Sink (audio_sink)
[2022-01-13 09:32:03.258] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-13 09:32:03.267] [error] Module 'bladerf_source' doesn't exist
[2022-01-13 09:32:03.267] [info] Initializing File Source (file_source)
[2022-01-13 09:32:03.277] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-13 09:32:03.286] [info] Initializing HackRF Source (hackrf_source)
[2022-01-13 09:32:03.303] [info] Initializing LimeSDR Source (limesdr_source)
[2022-01-13 09:32:03.332] [info] Initializing Network Sink (network_sink)
[2022-01-13 09:32:03.340] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-13 09:32:03.349] [info] Initializing RFspace Source (rfspace_source)
[2022-01-13 09:32:03.362] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-01-13 09:32:03.378] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-13 09:32:03.387] [info] Initializing Radio (radio)
[2022-01-13 09:32:03.564] [info] FM IF Noise reduction set to 32 taps
mmap() failed: Cannot allocate memory
Failed to create permanent mapping for memfd region with ID = 108523676
Ignoring received block reference with non-registered memfd ID = 108523676
[2022-01-13 09:32:03.581] [error] Could not open audio device
[2022-01-13 09:32:03.581] [info] Initializing Recorder (recorder)
[2022-01-13 09:32:03.596] [info] Initializing Rigctl Server (rigctl_server)
[2022-01-13 09:32:03.608] [info] Initializing SDRplay Source (sdrplay_source)
[2022-01-13 09:32:03.617] [error] Module 'sdrplay_source' doesn't exist
[2022-01-13 09:32:03.617] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
[ERROR] SoapySDR::Device::enumerate(remote) Resource temporarily unavailable

Are there any known successful installations on Pi3B?

good-point-lab, 2022-01-14:
> Hello,
> You're running out of ram. Try closing as much of other software as
> possible and, as a last resort, edit the config to remote module instances
> you don't need

Hi Alexandre,
Thank you for the prompt reply!   Pi 3B has 1 Gig, which seems not
sufficient to run SDR++...  I have seen a detailed example of success on
Pi4, but it has 2 Gigs.  By any chance, is a lighter version of the
software in your plans?

Alex Greysukh

AlexandreRouma, 2022-01-14:
There is no way to make the software lighter unfortunately. It's already very lightweight for it's feature set and CPU efficiency (some memory efficiency is sacrified for much higher speed).

I had tested SDR++ on a pi3 a while back and it ran. So I really suggest you try manually editing to the config to disable as many unused modules. Also drop the fftSize setting to 1024 (lowest it will go) and see if it start, if it does, raise it back up slowly  to the default of 65536 and see if it runs.

good-point-lab, 2022-01-14:
Alexandre,
I will try your suggestions and let you know.
Thanks!
Alex

good-point-lab, 2022-01-14:
Starting from scratch, set all the options from the Module List to OFF for cmake and fftSize =1024 in core/src/gui/main_window.h.  Still failed with the same 'cannot allocate memory' messages.  Also, listed the modules that were built:

 $ find . | grep '\.so' | sed 's/^/"/' | sed 's/$/",/' | sed '/sdrpp_core.so/d'
"./source_modules/rfspace_source/rfspace_source.so",
"./core/libcorrect/lib/libcorrect.so",

(Just for test, I installed gqrx and it runs)

I hope to see at least the SDR++ window...

AlexandreRouma, 2022-01-14:
setting fftSize in the code is useless. it gets overwritten from the config.
You have to set it in the created config.

Same thing for disabling building modules, if you had already installed and didn't uninstall before doing sudo make install they will still be loaded.

That's why I said to change that stuff through the config (`~/.config/sdrpp/config.json`)

good-point-lab, 2022-01-17:
I adjusted FFT size starting from 1024 and did not load modules, to begin with.  I do not see errors, but the UI does not show up at all.  (I selected "GL Driver, Full KMS" in Pi configuration and build glfw locally):

pi@raspberrypi:~/SDRPlusPlus $ sdrpp
[2022-01-17 18:57:33.736] [info] SDR++ v1.0.5
[2022-01-17 18:57:33.737] [info] Loading config
[2022-01-17 18:57:33.738] [warning] ConfigManager locked, waiting...
[2022-01-17 18:57:34.462] [info] Using OpenGL 3.0
[2022-01-17 18:57:35.647] [info] Loading icons
[2022-01-17 18:57:35.728] [info] Loading band plans
[2022-01-17 18:57:35.751] [info] Loading band plans color table
[2022-01-17 18:57:35.780] [info] Loading modules
[2022-01-17 18:57:35.781] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-01-17 18:57:35.797] [info] Loading color maps
[2022-01-17 18:57:35.804] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2022-01-17 18:57:35.813] [info] Loading /usr/share/sdrpp/colormaps/classic_green.json
[2022-01-17 18:57:35.820] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2022-01-17 18:57:35.828] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2022-01-17 18:57:35.836] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2022-01-17 18:57:35.844] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2022-01-17 18:57:35.852] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2022-01-17 18:57:35.860] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2022-01-17 18:57:35.868] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2022-01-17 18:57:35.876] [info] Loading /usr/share/sdrpp/colormaps/gqrx.json
[2022-01-17 18:57:35.884] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2022-01-17 18:57:36.003] [info] New DSP samplerate: 8000000.0 (source samplerate is 8000000.0)
[2022-01-17 18:57:36.004] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-01-17 18:57:36.121] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-01-17 18:57:36.123] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-01-17 18:57:36.144] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-01-17 18:57:36.152] [info] Ready.

AlexandreRouma, 2022-09-13:
The latest commits should have lowered the memory usage, please reopen if not solved.

closed 2022-09-13 by AlexandreRouma