AlexandreRouma/SDRPlusPlus #589: Segmentation fault (core dumped) - Linux Mint 20.2

SheetLightning, 2022-01-15:
I just updated my sources (15 Jan 2022) and re-built the application. I now get:

Segmentation fault (core dumped)

I had no problems during the cmake and make steps so I am not sure why I have a problem when running the application. I do see a number of 'error' messages for modules that I have turned off, for example:

[2022-01-15 16:28:08.607] [error] Module 'airspy_source' doesn't exist

I have an SDRPlay RSD Duo and an RTL SDR and have only enabled those particular modules to be compiled in so I guess it is not unexpected to see such errors. However it then ends with the segmentation fault. My SoapySDR and SoapySDRPlay libraries are current.

AlexandreRouma, 2022-01-15:
Please post the full logs, the error you posted is harmless and doesn't give any information of what's actually going on.

SheetLightning, 2022-01-15:
$ ./sdrpp
[2022-01-15 16:44:52.564] [info] SDR++ v1.0.5
[2022-01-15 16:44:52.564] [info] Loading config
[2022-01-15 16:44:52.678] [info] Using OpenGL 3.0
[2022-01-15 16:44:52.801] [info] Loading icons
[2022-01-15 16:44:52.807] [info] Loading band plans
[2022-01-15 16:44:52.809] [info] Loading band plans color table
[2022-01-15 16:44:52.815] [info] Loading modules
[2022-01-15 16:44:52.815] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-01-15 16:44:52.816] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-01-15 16:44:52.816] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-01-15 16:44:52.823] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-01-15 16:44:52.824] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-01-15 16:44:52.835] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-01-15 16:44:52.853] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-01-15 16:44:52.869] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-01-15 16:44:52.885] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-01-15 16:44:52.902] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-01-15 16:44:52.919] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-01-15 16:44:52.936] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-01-15 16:44:52.953] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-01-15 16:44:52.974] [info] Initializing Airspy Source (airspy_source)
[2022-01-15 16:44:52.985] [error] Module 'airspy_source' doesn't exist
[2022-01-15 16:44:52.985] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-15 16:44:53.001] [error] Module 'airspyhf_source' doesn't exist
[2022-01-15 16:44:53.001] [info] Initializing Audio Sink (audio_sink)
[2022-01-15 16:44:53.018] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-15 16:44:53.035] [error] Module 'bladerf_source' doesn't exist
[2022-01-15 16:44:53.035] [info] Initializing File Source (file_source)
[2022-01-15 16:44:53.052] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-15 16:44:53.068] [info] Initializing HackRF Source (hackrf_source)
[2022-01-15 16:44:53.085] [error] Module 'hackrf_source' doesn't exist
[2022-01-15 16:44:53.085] [info] Initializing LimeSDR Source (limesdr_source)
[2022-01-15 16:44:53.101] [error] Module 'limesdr_source' doesn't exist
[2022-01-15 16:44:53.101] [info] Initializing Network Sink (network_sink)
[2022-01-15 16:44:53.118] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-15 16:44:53.135] [error] Module 'plutosdr_source' doesn't exist
[2022-01-15 16:44:53.135] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-01-15 16:44:53.163] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-15 16:44:53.168] [info] Initializing Radio (radio)
Segmentation fault (core dumped)

AlexandreRouma, 2022-01-15:
I suggest you try resetting the radio configuration: `rm ~/.config/sdrpp/radio_config.json`

SheetLightning, 2022-01-15:
Unfortunately that did not help either.

SheetLightning, 2022-01-16:
I have attached an strace log if it helps. I tried disabling various modules in an effort to isolate which one might be causing the problem, but had no success unfortunately.

[sdrpp.log](https://github.com/AlexandreRouma/SDRPlusPlus/files/7877900/sdrpp.log)

drfruct, 2022-01-20:
The same issue when building on MacOS

./sdrpp
[2022-01-20 17:26:35.491] [info] SDR++ v1.0.5
[2022-01-20 17:26:35.492] [info] Loading config
[2022-01-20 17:26:35.492] [warning] ConfigManager locked, waiting...
[2022-01-20 17:26:35.829] [info] Loading icons
[2022-01-20 17:26:35.850] [info] Loading band plans
[2022-01-20 17:26:35.869] [info] Loading band plans color table
[2022-01-20 17:26:35.882] [info] Loading modules
[2022-01-20 17:26:35.883] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_sdr_source.dylib
[2022-01-20 17:26:35.906] [info] Loading /usr/local/lib/sdrpp/plugins/meteor_demodulator.dylib
[2022-01-20 17:26:35.919] [info] Loading /usr/local/lib/sdrpp/plugins/airspy_source.dylib
[2022-01-20 17:26:35.936] [info] Loading /usr/local/lib/sdrpp/plugins/network_sink.dylib
[2022-01-20 17:26:35.950] [info] Loading /usr/local/lib/sdrpp/plugins/sdrplay_source.dylib
[2022-01-20 17:26:35.965] [info] Loading /usr/local/lib/sdrpp/plugins/radio.dylib
[2022-01-20 17:26:35.980] [info] Loading /usr/local/lib/sdrpp/plugins/rigctl_server.dylib
[2022-01-20 17:26:35.999] [info] Loading /usr/local/lib/sdrpp/plugins/audio_sink.dylib
[2022-01-20 17:26:36.014] [info] Loading /usr/local/lib/sdrpp/plugins/m17_decoder.dylib
[2022-01-20 17:26:36.030] [info] Loading /usr/local/lib/sdrpp/plugins/frequency_manager.dylib
[2022-01-20 17:26:36.049] [info] Loading /usr/local/lib/sdrpp/plugins/recorder.dylib
[2022-01-20 17:26:36.066] [info] Loading /usr/local/lib/sdrpp/plugins/file_source.dylib
[2022-01-20 17:26:36.101] [info] Loading /usr/local/lib/sdrpp/plugins/airspyhf_source.dylib
[2022-01-20 17:26:36.112] [info] Loading /usr/local/lib/sdrpp/plugins/spyserver_source.dylib
[2022-01-20 17:26:36.128] [info] Loading /usr/local/lib/sdrpp/plugins/new_portaudio_sink.dylib
[2022-01-20 17:26:36.151] [info] Loading /usr/local/lib/sdrpp/plugins/discord_integration.dylib
[2022-01-20 17:26:36.162] [info] Loading /usr/local/lib/sdrpp/plugins/weather_sat_decoder.dylib
[2022-01-20 17:26:36.177] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_tcp_source.dylib
[2022-01-20 17:26:36.194] [info] Loading /usr/local/lib/sdrpp/plugins/hackrf_source.dylib
[2022-01-20 17:26:36.212] [info] Initializing Airspy Source (airspy_source)
[2022-01-20 17:26:36.261] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-20 17:26:36.278] [info] Initializing Audio Sink (audio_sink)
[2022-01-20 17:26:36.303] [info] Initializing BladeRF Source (bladerf_source)
[2022-01-20 17:26:36.303] [error] Module 'bladerf_source' doesn't exist
[2022-01-20 17:26:36.303] [info] Initializing File Source (file_source)
[2022-01-20 17:26:36.306] [info] Initializing Frequency Manager (frequency_manager)
[2022-01-20 17:26:36.322] [info] Initializing HackRF Source (hackrf_source)
[2022-01-20 17:26:36.352] [info] Initializing LimeSDR Source (limesdr_source)
[2022-01-20 17:26:36.357] [error] Module 'limesdr_source' doesn't exist
[2022-01-20 17:26:36.357] [info] Initializing Network Sink (network_sink)
[2022-01-20 17:26:36.372] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-20 17:26:36.391] [error] Module 'plutosdr_source' doesn't exist
[2022-01-20 17:26:36.391] [info] Initializing RFspace Source (rfspace_source)
[2022-01-20 17:26:36.405] [error] Module 'rfspace_source' doesn't exist
[2022-01-20 17:26:36.405] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2022-01-20 17:26:36.927] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-20 17:26:36.929] [info] Initializing Radio (radio)
[1]    31632 segmentation fault  ./sdrpp

Luceoria, 2022-01-25:
Similar fail on the most recent source. Kernel logs show the following:
```
[17881.981138] traps: sdrpp[9029] general protection fault ip:7f6fd06f56b3 sp:7fff8506aa00 error:0 in radio.so[7f6fd06c5000+53000]
[17887.242002] traps: sdrpp[9068] general protection fault ip:7f080cfc56b3 sp:7ffe13df9260 error:0 in radio.so[7f080cf95000+53000]
[17903.784256] traps: sdrpp[9109] general protection fault ip:7f416bfc06b3 sp:7fff917c7a30 error:0 in radio.so[7f416bf90000+53000]
[18136.610126] traps: sdrpp[9199] general protection fault ip:7f0cc10856b3 sp:7ffd76d35b30 error:0 in radio.so[7f0cc1055000+53000]
```

rs1729, 2022-01-30:
Just got the same (I guess) on ubuntu 20.04

> ./sdrpp
[...]
[2022-01-30 15:58:29.018] [info] Initializing Radio (radio)
Segmentation fault (core dumped)

Last build was from November. Commit 5a198290684021b8080dc02531555f94098ee88b was ok, next commit b141c4b2a50962df8c8359018a5b9489166469b5 produced the `Segmentation fault (core dumped)`.
Made a debug version:

> [2022-01-30 16:08:03.074] [info] Initializing Radio (radio)
[New Thread ...]
Thread 1 "sdrpp" received signal SIGSEGV, Segmentation fault.
0x00007fffb3ce06b3 in WFMDemodulator::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, VFOManager::VFO*, float, float, ConfigManager*) () from /usr/lib/sdrpp/plugins/radio.so

Looking at the log output

> [2022-01-30 17:22:52.129] [info] Loading /usr/lib/sdrpp/plugins/radio.so

(!) so it was loading the plugins from the older installation, which is not compatible with the new built functions ...
I guess that's why the readme.md says, when running from `build`for development

> Create a new root directory
[...]

However, the instructions are a bit confusing.
It starts in the top directory,
`./create_root.sh` does `cp -r root root_dev`, so it copies `root` into `root_dev`, but later it is recommended:
```
"modulesDirectory": "./root_dev/modules",
"resourcesDirectory": "./root_dev/res",
```
So either `cp -r root/* root_dev` or change `config.py`
```
"modulesDirectory": "./root_dev/root/modules",
"resourcesDirectory": "./root_dev/root/res",
```
Now in the build directory (if you run `./sdrpp -r ../root_dev/` from the build directory the first time to generate a default config, you get the segfault ...), but then the changes in the `config.json` are relative to the top directory, which is later explained that you can also "Or, if you wish to run from the build directory, you will need to correct the directories in the config.json file" as you find out as soon as it goes wrong.

Anyway, without the modules it starts now. But if you copy the `find | sed` output to `"modules": []` and don't remove the last comma, it will reset the `config.py`
```
[2022-01-30 17:22:51.284] [info] Loading config
[2022-01-30 17:22:51.286] [error] Config file '/home/.../build/SDRPlusPlus/./root_dev//config.json' is corrupted, resetting it
```
and you can start all over again... ;)

Now I can test the new build.
Installing `sudo make install` before testing would replace the /usr/lib-files and solve the problem as well, and there would be no segfault.

SheetLightning, 2022-03-10:
Just pulled the latest version and got the same result, then re-read the detailed post by rs1729. After this I simply did:

sudo rm -R /usr/lib/sdrpp
rm -R ~/.config/sdrpp

It now starts. It was reading the previous version of the libraries in /usr/lib/sdrpp as observed.

closed 2022-08-17 by aosync