AlexandreRouma/SDRPlusPlus #592: Initializing Recorder (recorder)

zainkamoi, 2022-01-16:
Have both tried the latest prebuilt release 1.0.4 (sdrpp_ubuntu_hirsute_amd64.deb) , and built from scratch with default options but both builds run into same issue

> 
 [2022-01-16 21:37:35.914] [info] SDR++ v1.0.5
[2022-01-16 21:37:35.914] [info] Loading config
[2022-01-16 21:37:35.914] [info] Missing key in config bandPlanPos, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config decimationPower, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fastFFT, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fftRate, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fftSize, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fftWindow, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fullWaterfallUpdate, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config fullscreen, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config iqCorrection, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config menuElements, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config offsetMode, repairing
[2022-01-16 21:37:35.914] [warning] ConfigManager locked, waiting...
[2022-01-16 21:37:35.914] [info] Missing key in config showMenu, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config theme, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config vfoColors, repairing
[2022-01-16 21:37:35.914] [info] Missing key in config vfoOffsets, repairing
[2022-01-16 21:37:35.915] [info] Unused key in config menuOrder, repairing
[2022-01-16 21:37:36.067] [info] Using OpenGL 3.0
[2022-01-16 21:37:36.172] [info] Loading icons
[2022-01-16 21:37:36.179] [info] Loading band plans
[2022-01-16 21:37:36.182] [info] Loading band plans color table
[2022-01-16 21:37:36.187] [info] Loading modules
[2022-01-16 21:37:36.187] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-01-16 21:37:36.189] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-01-16 21:37:36.190] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-01-16 21:37:36.204] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-01-16 21:37:36.220] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-01-16 21:37:36.237] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-01-16 21:37:36.253] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-01-16 21:37:36.271] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2022-01-16 21:37:36.290] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-01-16 21:37:36.306] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-01-16 21:37:36.322] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2022-01-16 21:37:36.341] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-01-16 21:37:36.355] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-01-16 21:37:36.375] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2022-01-16 21:37:36.391] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-01-16 21:37:36.405] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-01-16 21:37:36.424] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-01-16 21:37:36.438] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-01-16 21:37:36.458] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-01-16 21:37:36.471] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2022-01-16 21:37:36.471] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-01-16 21:37:36.489] [warning] ConfigManager locked, waiting...
[2022-01-16 21:37:36.489] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2022-01-16 21:37:36.506] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-01-16 21:37:36.530] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-01-16 21:37:36.539] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-01-16 21:37:36.556] [info] Initializing Airspy Source (airspy_source)
[2022-01-16 21:37:36.578] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-01-16 21:37:36.594] [info] Initializing Audio Sink (audio_sink)
[2022-01-16 21:37:36.603] [info] Initializing HackRF Source (hackrf_source)
[2022-01-16 21:37:36.627] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-01-16 21:37:36.637] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-01-16 21:37:36.653] [info] Initializing Radio (radio)
[2022-01-16 21:37:36.728] [info] FM IF Noise reduction set to 32 taps
**[2022-01-16 21:37:36.728] [info] Initializing Recorder (recorder)
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be number, but is null
Aborted (core dumped)**
>

Earlier i run an rather older version of sdrpp, 0.0.25, before rebuilding, and this version had no issues like this.

AlexandreRouma, 2022-01-16:
it's normal, you're updating from a pre-alpha version.
delete the  `~/.config/sdrpp` folder completely

zainkamoi, 2022-01-16:
Wow, thank you, This worked!

closed 2022-01-16 by AlexandreRouma