AlexandreRouma/SDRPlusPlus #599: Running #567 and #568 in MacOS app crashes

ja-Bertolin, 2022-01-21:
Currently I am running #562 and works like a charm

I have just installed 567 and 568 and both provokes a crash of the app, below the infomation provided, just in case it helps.

Process:               sdrpp [78165]
Path:                  /private/var/folders/*/SDR++.app/Contents/MacOS/sdrpp
Identifier:            oorg.sdrpp.sdrpp
Version:               1.0.5
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           sdrpp [78165]
User ID:               503

Date/Time:             2022-01-21 23:23:02.841 +0100
OS Version:            macOS 11.6.2 (20G314)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        4C0C046A-69D9-463D-E94F-F465EFFD427F

Time Awake Since Boot: 1800 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [78165]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.IOSurface           	0x00007fff2885404c IOSurfaceClientDecrementUseCount + 22
1   com.apple.QuartzCore          	0x00007fff2718c60e release_image(_CAImageQueue*, unsigned long, CAImageQueueReleased*, bool) + 95
2   com.apple.QuartzCore          	0x00007fff2715d2eb CAImageQueueCollect_ + 196
3   com.apple.AppKit              	0x00007fff23c3c556 -[NSCGLSurface flushRect:] + 72
4   com.apple.AppKit              	0x00007fff23c3c490 NSCGLSurfaceFlush + 214
5   GLEngine                      	0x00007fff6c01faea glSwap_Exec + 186
6   com.apple.opengl              	0x00007fff6bde52eb CGLFlushDrawable + 59
7   com.apple.AppKit              	0x00007fff2355c5f0 -[NSOpenGLContext flushBuffer] + 20
8   libglfw.3.dylib               	0x000000010e5c6405 swapBuffersNSGL + 221
9   libsdrpp_core.dylib           	0x000000010dc71e4d LoadingScreen::show(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) + 477
10  libsdrpp_core.dylib           	0x000000010dca3fb3 MainWindow::init() + 5379
11  libsdrpp_core.dylib           	0x000000010dc65600 sdrpp_main(int, char**) + 22704
12  libdyld.dylib                 	0x00007fff20902f3d start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff208b4bb2 __semwait_signal + 10
1   libsystem_c.dylib             	0x00007fff20834c1a nanosleep + 196
2   libc++.1.dylib                	0x00007fff2088793a std::__1::this_thread::sleep_for(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > const&) + 73
3   libsdrpp_core.dylib           	0x000000010dc1753e ConfigManager::autoSaveWorker() + 174
4   libsdrpp_core.dylib           	0x000000010dc5d21e void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (ConfigManager::*)(), ConfigManager*> >(void*) + 62
5   libsystem_pthread.dylib       	0x00007fff208e78fc _pthread_start + 224
6   libsystem_pthread.dylib       	0x00007fff208e3443 thread_start + 15

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff208e3420 start_wqthread + 0

Thread 3:: Dispatch queue: com.apple.root.background-qos
0   libsystem_kernel.dylib        	0x00007fff208b2cb6 close + 10
1   com.apple.AppKit              	0x00007fff231ecbc3 -[NSPersistentUICrashHandler inspectCrashDataWithModification:handler:] + 738
2   com.apple.AppKit              	0x00007fff23365c5c __82-[NSPersistentUIRestorer tearDownStateRestorationApparatusAndResumeWindowOrdering]_block_invoke + 47
3   com.apple.AppKit              	0x00007fff23d0067d ___NSPersistentUIDispatchQueueAsync_block_invoke + 28
4   libdispatch.dylib             	0x00007fff2073c623 _dispatch_call_block_and_release + 12
5   libdispatch.dylib             	0x00007fff2073d806 _dispatch_client_callout + 8
6   libdispatch.dylib             	0x00007fff2074c976 _dispatch_root_queue_drain + 676
[... 465 lines cut ...]
mapped file                      229.2M       19 
shared memory                      788K       16 
===========                     =======  ======= 
TOTAL                              2.1G     1130 
TOTAL, minus reserved VM space     1.4G     1130 

Model: MacBookPro13,3, BootROM 447.60.3.0.0, 4 processors, Quad-Core Intel Core i7, 2.6 GHz, 16 GB, SMC 2.38f12
Graphics: kHW_IntelHDGraphics530Item, Intel HD Graphics 530, spdisplays_builtin
Graphics: kHW_AMDRadeonPro450Item, AMD Radeon Pro 450, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.9)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Belkin USB-C LAN, Ethernet, en7
Network Service: USB 10/100/1000 LAN, Ethernet, en10
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: Ugreen Storage Device
USB Device: USB 3.0 Bus
USB Device: AIRSPY HF+
USB Device: Arduino Leonardo
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2

AlexandreRouma, 2022-01-21:
Can't replicate and the crash dump seems to indicate an OpenGL issue. (nothing was changed in the opengl code between the two versions you mention, especially at the exact location of the crash)

Can you run the executable inside the .app from a terminal and copy the output here

AlexandreRouma, 2022-01-22:
after updating glfw I can replicate the issue, it seems like the version shipped in the .app is broken

AlexandreRouma, 2022-01-22:
After trying all three versions you mentioned, they all crash inside glfw like yours did. The weird thing is GLFW hasn't been updated in a while so I can only conclude the issue was a recent MacOS update.

ja-Bertolin, 2022-01-22:
Tested #577 and it works fine, thanks for the effort fixing it

AlexandreRouma, 2022-01-22:
I think it might be an intermittent issue, please keep an eye out and let me know if it breaks again

AlexandreRouma, 2022-10-21:
closing as probably was a glfw issue that is out of my control