AlexandreRouma/SDRPlusPlus #600: macOs IF AGC window expand

loewal, 2022-01-22:
Hi Alexandre,

When I click "Parameters" behind "IF AGC", the popup-window slowly expands to full 
[image]
width of the current window.
After that I can't even press stop or play, because that same window appears then.

Grtz.
Macos Monterey

pbij2001, 2022-01-22:
Where do I find these AGC parameters? Or are they HW specific?

AlexandreRouma, 2022-01-22:
Well that's one strange issue, I'll try to replicate on the remote M1 mac I have access to and let you know

loewal, 2022-01-22:
Intel mac

AlexandreRouma, 2022-01-22:
It does the same on M1.
I suspect it's an ImGui bug on MacOS, so I'll see if an update to imgui helps.

AlexandreRouma, 2022-01-22:
@pbij2001 they're the AGC parameters for the SDRplay source. so yes it's hardware specific.,+

pbij2001, 2022-01-23:
Tnx. Understood. The top 4 sliders of this settings window would be great for SSB/CW/AM audio AGC.

AlexandreRouma, 2022-01-23:
It directly controls the SDRplaay API, I didn't implement it

pbij2001, 2022-01-23:
Clear. Back to waiting mode for general AGC. Hope you can "DSP model" the TS850 one day in that area :-))