AlexandreRouma/SDRPlusPlus #601: rigtcl server does not work after the latest changes

dl7vea, 2022-01-22:
The rigctl server does not receive neither transmit frequency changes after the last changes

AlexandreRouma, 2022-01-22:
Nothing was changed in the rigctl server, I suspect this is an issue with the latest himlib update that completely breaks the protocol.

dl7vea, 2022-01-22:
I'm using the old hamlib and nothing was changed in this area on my side.

So it might be indeed the network library.

dl7vea, 2022-01-22:
If I try to connect sdrpp rigctl via:

rigctl -m 2 -r 127.0.0.1:4534

I see a short while status "Connected" in sdrpp, then a "Listening" again and the command line tool reports Communication timed out

dl7vea, 2022-01-22:
With the release a view minutes ago it works again!

Thank you!

closed 2022-01-22 by dl7vea