AlexandreRouma/SDRPlusPlus #602: Unable to load plugins

carbform, 2022-01-22:
After installing sdrpp from the deb package on Ubuntu 21.10 and on opening from terminal, I get these 5 errors.
````````````````````````````````````````````````
[error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[error] Couldn't load /usr/lib/sdrpp/plugins/soapy_source.so.
[error] Could not list devices

[error] Module 'sdrplay_source' doesn't exist
[error] Module 'soapy_source' doesn't exist

AlexandreRouma, 2022-01-22:
Deb files do not contain dependencies. If you want to use those modules you need to have their needed dependencies installed. In this cause, the SDRplay API and SoapySDR.

Note that these errors are only really warnings and can be ignored since they do not cause problems.

carbform, 2022-01-25:
I installed sdrpp using the deb package and it worked perfectly with a DVB-T stick. I could listen to all FM radio stations in my area. Since then, I uninstalled it and reinstalled it again. This time, however, I am unable to pick up any FM signals. The DVB-T dongle is working fine (used rtl_test) and I made sure that no dump1090 and dump1090-mutability services were running when I opened sdrpp.

closed 2022-01-25 by carbform