AlexandreRouma/SDRPlusPlus #604: error while loading shared libraries: libGLEW.so.2.2

kixell, 2022-01-23:
I'm having a problem when launching sdrpp I get the following error

`sdrpp: error while loading shared libraries: libGLEW.so.2.2: cannot open shared object file: No such file or directory`

I ran the following 

sudo apt install libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev
wget https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.4/sdrpp_debian_sid_amd64.deb
sudo dpkg -i sdrpp_debian_sid_amd64.deb

sdrpp
`sdrpp: error while loading shared libraries: libGLEW.so.2.2: cannot open shared object file: No such file or directory`

uname -a
`Linux kevin-NV52-Series 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

I saw on another [similar issue](https://github.com/AlexandreRouma/SDRPlusPlus/issues/87) that not having a supported version of OpenGL could cause this problem.
glxinfo |grep "OpenGL version"
OpenGL version string: 3.0 Mesa 21.0.3

Not sure what I did wrong. I'm not the strongest Linux user.

kistlin, 2022-01-23:
Maybe you have another version?
Do you find something with `find /usr/lib -iname "libGLEW*"`?

AlexandreRouma, 2022-01-23:
Probably installed the deb for the wrong distro version.

kixell, 2022-01-23:
> Probably installed the deb for the wrong distro version.

This was the problem, it's working thanks.

closed 2022-01-23 by kixell

delboy85341, 2022-09-21:
I upgraded from Ubuntu 20.04 to 22.04 then SDRpp didn't work.  I  went to sdrpp.org, clicked releases and downloaded and installed the hirsute version since there was no jammy version.  That didn't work either.I looked around and found a forum where someone suggested downloading the nightly build that there was a jammy version there.  I tried   nightly builds and found the names of the various  files available at the bottom of the page, but they aren't links.  Does anyone know where I can download the nightly builds?  thanks_

AlexandreRouma, 2022-09-21:
get the nightly https://www.sdrpp.org/nightly

delboy85341, 2022-09-22:
> get the nightly https://www.sdrpp.org/nightly

The links weren't there yesterday but they're there now. thanks