AlexandreRouma/SDRPlusPlus #605: libsdrpp_core.so is missing _INFO_ symbol

Dantali0n, 2022-01-23:
I get an error that terminates sdrplusplus during launch saying `libsdrpp_core.so is missing _INFO_ symbol`.

I have used cmake and make and the compilations went smoothly. Using ldd I don't see any linked libraries missing.

```
erminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be array, but is null
Aborted (core dumped)
```

gvanem, 2022-01-23:
> libsdrpp_core.so is missing _INFO_ symbol.

Why should it export a `_INFO_` symbol? Only *true* modules (like the `rtl_sdr_source.dll` plugin) should do that.

AlexandreRouma, 2022-01-23:
I suspect you're trying to load the core lib as a module?????
review your configuration.

Dantali0n, 2022-01-23:
> I suspect you're trying to load the core lib as a module????? review your configuration.

Indeed this file was present in `modulesDirectory` although not defined in `modules` in the config.json. Seems that all .so files in the modules directory are loaded regardless if they are defined in `modules`. Is this by design?

Similarly I would like to cease this opportunity to suggest, Perhaps it is better to not delete the contents of config.json if its invalid. I think it makes initial configuration more painful than is necessary.

AlexandreRouma, 2022-01-23:
Yes. That is the point of the `modules` directory and how modules are loaded in an installed version.

The contents of the config files are reset when invalid because for normal users this is the best option. Developers are expected to make backups if they are going to edit config files and make mistakes.

closed 2022-01-23 by Dantali0n