AlexandreRouma/SDRPlusPlus #606: Frequency doesn't save/reload correctly

Rua, 2022-01-23:
With center tuning enabled, if I tune to a certain frequency, say 90.9 MHz, and then exit, the frequency is correctly saved in config.json. But if I then start SDR++ back up again, the frequency is slightly off, 90.904 in my case. It's not restoring the saved frequency correctly.

With center tuning disabled, config.json does not even contain the correct frequency. If I center at 90.9, then tune down to 90.7, and exit, config.json will contain 90.9 instead of 90.7, so it seems to be saving the center frequency rather than the tuned frequency. Reopening SDR++ will then put it on 90.904 as above.

AlexandreRouma, 2022-01-23:
the frequency saved in the config.json is NOT the frequency your VFO is at. It's the center frequency.
THe offset of each VFO is stored separately.
The frequency soetimes being off after a restart is a known issue

Rua, 2022-01-23:
Then it seems that it's not restoring the VFO frequency if center tuning is disabled, but rather setting the VFO to the center frequency.

AlexandreRouma, 2022-01-23:
> The frequency soetimes being off after a restart is a known issue
with some sources they start out at a lower samplerate which causes the vfo manager to put the VFOs back in view

closed 2022-09-29 by AlexandreRouma