AlexandreRouma/SDRPlusPlus #607: IQ-correction and Audio

gvanem, 2022-01-23:
The below audio-recording of the local Airport VOR (at 125.250 MHz, AM, 5kHz b/w) shows an puzzling issue
https://www.watt-32.net/dx/Flesland-Airport-VOR-125.250MHz.mp3

* With IQ-correction off and tuning in small steps up and down, there is an annoying audio *beat-tone*.
between 00:03 and 00:14 in the recording.
* With IQ-correction on, there is no suck *beat-tone*. E.g. from 00:15 onwards.

Must be an issue in SDR++ since I've never heard such beat-tone in SDRuno, SDR#, GQrx etc.
Can someone please explain this?

gvanem, 2022-01-23:
Forgot to say I was in "*center tuning mode*". Any relation to my issue?

AlexandreRouma, 2022-01-23:
that is just expected behavior from DC correction. You're nuking the DC component of the signal, it WILL mess with a centered AM signal.

gvanem, 2022-01-23:
> ... from DC correction. 

I'm even more confused now. **IQ-correction** vs. **DC correction**; I fail to see the relationship.

> You're nuking the DC component of the signal,

I did? Care to explain in a bit more detail?

I thought also it was an *Beat Frequency Oscillator* feature.

AlexandreRouma, 2022-01-23:
IQ correction contains a DC correction algorithm. The goal is to remove the DC offset of the signal to remove the "spike" you see in the center of the IF in some SDRs (RTL-SDR/hackrf & co). This is usueful because the DC spike can mess with some modulations like FM or PSK and correcting it means being able to receive those signals right in the center of the IF.

With AM, the demodulation relies on the carrier. If you're in center tuning mode, guess what, it'll will be exactly in the center of the IF or close enough to be picked up by the DC correction and (partially or fully) removed. After that, it's like trying to receive a DSB signal with an AM demodulator. The demodulator and AGC get confused at the lack of carrier and that's what you're hearing.

pbij2001, 2022-01-23:
Airspy solved this by always having an offset of 5kHz between the demod signal and IQ center, even in center tuning mode. This avoids any carrier cancellation issues. Especially important in narrow CW, AM(S) and FM. Not sure if this is hard coded in the HW FW or the SW.

AlexandreRouma, 2022-01-23:
it's the way the IF is sampled. Cheap quadrature RF sampling has this issue. Direct sampling like the airspy does doesn't have that issue.

pbij2001, 2022-01-23:
This faint "RF hole" is what you see on Discovery, independent of frequency. The IQ center is shifted 5kHz
[image]

AlexandreRouma, 2022-01-23:
Anyway I think this is resolved so I'll close the issue

closed 2022-01-23 by AlexandreRouma

gvanem, 2022-01-24:
>  The goal is to remove the DC offset of the signal to remove the "spike" you see ...

Many thanks for the insightful explanation.