AlexandreRouma/SDRPlusPlus #613: Menu does not show on Linux Debian 11

dl7vea, 2022-01-25:
Running the latest version. Build on a Debian 11 with a NVidia graphics adapter. 

The menu on the right side is empty. From time to time it flickers very short and then disappears. So I'm not able to control the software. It runs once I started the SDR, but completely unmanageable. 

Started sdrpp from a console window. Nothing particular expect the normal "info" messages.

Does anyone have had a similar experience?

AlexandreRouma, 2022-01-25:
Firstly, what do you mean by "latest version", can you send the exact version number and build date/time (shown in the window title and in the credits)
Secondly, it would be helpful to get a screen recording of the issue to see it happening.
Finally I'd suggest updating to the latest nightly build <https://www.sdrpp.org/nightly> and checking if the issue is still there, as well as updating your graphics drivers.

dl7vea, 2022-01-25:
I build from the sources of today. Version is 1.0.5 Build Jan 25, 2022, 13:17:25

I'll check how to do a screen recording. 

The Debian including all drivers is setup up yesterday with the latest drivers available and packages installed yesterday.

I'll check the nightly build now.

dl7vea, 2022-01-25:
The nightly build shows v1.0.5 with build date Jan 24, 2022 00:03:44.

And shows the same behavior.

dl7vea, 2022-01-25:
[image]
[image]

The most time the area is empty, but from time to time scrolling and clicking around it shows like in the second picture

AlexandreRouma, 2022-01-25:
to me that looks like a broken graphics driver. What hardware are you using?

dl7vea, 2022-01-25:
Alexandre, thank you for your answer.

So I didn't found a newer driver. And I didn't found a way to setup the driver in a different way. And have no more clue where to search. So for the time being it's a pity, that it does not work this way.

closed 2022-01-25 by dl7vea

AlexandreRouma, 2022-01-25:
I'll let you know if I find out of a bug somewhere.