AlexandreRouma/SDRPlusPlus #618: Application hang and infinite loop using Pipewire

cloventt, 2022-01-28:
Hi all,

I have compiled SDR++ from current master (260651fb5c328346044ec6562f7d590c220dc203) on Fedora 35. This distro uses Pipewire for handling audio rather than pulseaudio.

I was able to get SDR++ to run properly at first. However when I selected my Bluetooth headphones in the audio sink options, the application froze and began spamming this error to console:

```
RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

RtApiCore::callbackEvent(): the JACK buffer size has changed ... cannot process!

```

The only intervention is to SIGTERM the application. I reset the config file back to defaults and was able to restart the application. However after another restart with no changes the error is now spammed on startup and the app remains hanged. I've tried resetting the  `~/.config/sdrpp/audio_sink_config.json` file to defaults, and editing it with different audio device names and sample rates, but it does not seem to help. 

The error log seems to be coming from the RtAudio lib at this line: https://github.com/thestk/rtaudio/blob/f9fd8726ccb688baa43aaae642179a69d353f6d2/RtAudio.cpp#L2730

AlexandreRouma, 2022-01-28:
Not a SDR++ issue, open an issue with the rtaudio team: https://github.com/thestk/rtaudio/issues

closed 2022-01-28 by AlexandreRouma

ericek111, 2022-03-03:
One workaround is to [fix PipeWire's buffer size for JACK](pw-metadata -n settings 0 clock.force-quantum 1024) to one value:
```
pw-metadata -n settings 0 clock.force-quantum 1024
```